NVIDIA/Isaac-GR00T #170: Any results on finetuning and evaluating Gr00t on SimplerEnv

phandat128, 2025-05-20:
Hi Team,
Thank you very much for sharing the work!
I recently have tried to test GR00T's performance in simulation environment. Following some paper experiments on SimplerEnv, i have finetuned GR00T models on fractal dataset for google-robot. But after 50k steps finetuning with batch_size=64, the robot still cannot do any success task. 
FYI, here is the fractal dataset i have used: https://huggingface.co/datasets/IPEC-COMMUNITY/fractal20220817_data_lerobot.
Here is an example video:
https://github.com/user-attachments/assets/4407366b-b34e-4fcc-b5e2-3ff8d2cb8f8f

I wonder that if anyone have a better finetuning and evaluating GR00T on SimplerEnv. Please provide me some information. 
Thank you all!

Catchip, 2025-06-05:
https://github.com/DelinQu/SimplerEnv-OpenVLA

LukeLIN-web, 2025-08-10:
> Hi Team, Thank you very much for sharing the work! I recently have tried to test GR00T's performance in simulation environment. Following some paper experiments on SimplerEnv, i have finetuned GR00T [...]

Did you make it success now?