NVIDIA/Isaac-GR00T #172: Some bug about torchvision_av backend

shenyanqing1105, 2025-05-21:
### 🐛 Describe the bug

when using the `video_backend = torchvision_av`, `get_frames_by_timestamps` function in `gr00t/utils/video.py` like this:
   ```
 elif video_backend == "torchvision_av":
        # set backend
        torchvision.set_video_backend("pyav")
        # set a video stream reader
        reader = torchvision.io.VideoReader(video_path, "video")
        # set the first and last requested timestamps
        # Note: previous timestamps are usually loaded, since we need to access the previous key frame
        first_ts = timestamps[0]
        last_ts = timestamps[-1]
        # access closest key frame of the first requested frame
        # Note: closest key frame timestamp is usally smaller than `first_ts` (e.g. key frame can be the first frame of the video)
        # for details on what `seek` is doing see: https://pyav.basswood-io.com/docs/stable/api/container.html?highlight=inputcontainer#av.container.InputContainer.seek
        reader.seek(first_ts, keyframes_only=True)
        # load all frames until last requested frame
        loaded_frames = []
        loaded_ts = []
        for frame in reader:
            current_ts = frame["pts"]
            loaded_frames.append(frame["data"])
            loaded_ts.append(current_ts)
            if current_ts >= last_ts:
                break
            if len(loaded_frames) >= len(timestamps):
                break
        reader.container.close()
        reader = None
        frames = np.array(loaded_frames)
        return frames.transpose(0, 2, 3, 1)
    else:
        raise NotImplementedError
```
Compared with the official LeRobot version, the time check was deleted. 

So, for example, if I want the frame of **32s**, the returned frame corresponds to **16s**. The time gap is huge....

### Versions

/

EricYang3721, 2025-06-29:
Thanks for pointing this out. I found a huge performance gap when I use `CachedLeRobotSingleDataset` and `LeRobotSingleDataset` when using the `torchvision_av` backened. For first one load all images at once, while the second one uses the function you pointed out.

youliangtan, 2025-09-19:
https://github.com/NVIDIA/Isaac-GR00T/pull/373

closed 2025-09-19 by youliangtan