NVIDIA/Isaac-GR00T #178: G1 deployment effect is not good

beyondli, 2025-05-21:
### 🚀 The feature, motivation and pitch

N1 deployment on G1, action did not accomplish

HomerIsAFool, 2025-06-05:
Hi, i am also deploying gr00t on G1.
i fintune the model with G1_Stack_Blocking dataset and the MSE is around 0.004. I am using only one camera on the head(although the dataset provide videos from 3 cameras )
When i deployint the model on unitree G1, i found that the robot can not reach out for the block on the table, and just shaking its arm slightly. 
What is the response of your robot after model deployment? Is it similar to mine?

youliangtan, 2025-06-16:
Can try with the new n1.5, along with new G1 config. Each G1 is configured differently with different tuning of gains etc, so ensure you collect your own data.

closed 2025-06-16 by youliangtan

mohdammari97, 2025-07-27:
Hello everyone. I want to fine-tune the model on my Unitree G1. How did you deploy the model on your Unitree G1?
Have you created a custom python script that takes the outputs from GR00T and use it in a script to control the robot via https://github.com/unitreerobotics/unitree_sdk2_python/ ?
Do you keep the robot running in debug mode and then run the model?
I'd appreciate it a lot if you could provide me details regarding the deployment.
Thank you!

mohdammari97, 2025-08-07:
> [@mohdammari97](https://github.com/mohdammari97) From what I know, you need to select debug mode on the unitree g1 user interface before that. Internally we run the policy as a server, then have an inference script that keep "query" the policy with `get_action()` asychronously. You can write your own script to implement this.

@youliangtan Thank you for your response.
I understood the debug mode part. However, I am not sure if I got the second part. Isn't the inference.py script used to run the model/policy?
It sounded as if they are two separate steps. Also, does the inference script have direct control to the robot? Does it directly write to the robot joints or should this be done as an extra step from my side?
I would appreciate it if you could provide a template or an example of your implementation.

mohdammari97, 2025-08-07:
> > Hello everyone. I want to fine-tune the model on my Unitree G1. How did you deploy the model on your Unitree G1? Have you created a custom python script that takes the outputs from GR00T and use it in a script to control the robot via https://github.com/unitreerobotics/unitree_sdk2_python/ ? Do you keep the robot running in debug mode and then run the model? I'd appreciate it a lot if you could provide me details regarding the deployment. Thank you!
> 
> Hava you has a specific step on how to simulate this model on Issac sim and Issac lab. I'd be appreciated since the step is not released.

@hz3014 I haven't run my model yet. I just started with this project and I am about to start collecting data for training.