NVIDIA/Isaac-GR00T #180: Request for DOF order for each hand (GR1)

eyalmentee, 2025-05-21:
Hi,
In gr1's modality.json, the coarse DoF ranges are listed as:

```
"left_hand": { "start": 7, "end": 13 },
"right_hand": { "start": 29, "end": 35 }
```
Could you please share the detailed DoF order for each hand? 

Thanks!

wyt1234, 2025-05-21:
7个臂关节：
肩部俯仰（Shoulder pitch）
肩部滚转（Shoulder roll）
肩部偏航（Shoulder yaw）
肘部俯仰（Elbow pitch）
腕部偏航（Wrist yaw）
腕部滚转（Wrist roll）
腕部俯仰（Wrist pitch）
对于每只手（左右手）：
6个手指关节：
小指（Little finger）
无名指（Ring finger）
中指（Middle finger）
食指（Index finger）
拇指旋转（Thumb rotation）
拇指弯曲（Thumb bending）
对于腰部：
3个关节：
躯干腰部偏航（torso waist yaw）
躯干腰部俯仰（torso waist pitch）
躯干腰部滚转（torso waist roll）

MaggieK410, 2025-06-25:
Hi, I have a question for the fingers. The fourier hand has a proximal and a distal joint but here only one is in the state space for GR00T. Does "Little finger" refer to proximal or distal joint?

maxvipon, 2025-07-07:
@wyt1234 could you please clarify which model of the GR-1 hands is used in training (and in your explanation)? Because the GR-1 comes with different hands (DexHands, InspireHands), and I see the same as @eyalmentee — for InspireHands, the thumb bends first, and then the rotation.

xwjiang2010, 2025-09-16:
> In gr1's modality.json, the coarse DoF ranges are listed as:
> 
> 
> 
> "left_hand": { "start": 7, "end": 13 },
> "right_hand": { "start": 29, "end": 35 }
> Could you please share the detailed DoF order for each hand?

I believe these are using InspireHands.
I will close this for now. Feel free to reopen if more issues.

closed 2025-09-16 by xwjiang2010