NVIDIA/Isaac-GR00T #189: How to create modality.json for AgiBotWorld-Alpha ?

zouchengzhuo, 2025-05-26:
### 📚 The doc issue

I can see the Schema for modality in https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/LeRobot_compatible_data_schema.md，but dont know how to create it

Catchip, 2025-05-31:
I finetuned at AgiBotWorld task 327, the data comes from [agibotworld-lerobot](https://huggingface.co/datasets/weijian-sun/agibotworld-lerobot)
A simple setting is as follows:
# modality.json
```json
{
  "state": {
    "observation.state": {
      "start": 0,
      "end": 20
    }
  },
  "action": {
    "action": {
      "start": 0,
      "end": 22
    }
  },
  "video": {
    "hand_left": {
      "original_key": "observation.images.hand_left"
    },
    "hand_right": {
      "original_key": "observation.images.hand_right"
    },
    "top_head": {
      "original_key": "observation.images.top_head"
    }
  },
  "annotation": {
    "human.task_description": {
      "original_key": "task_index"
    }
  }
}

```

# dataconfig
```python
class AgibotG1DataConfig(BaseDataConfig):
    # video_keys = [
        # "video.back_left_fisheye",
        # "video.back_right_fisheye",
        # "video.head_left_fisheye",
        # "video.head_right_fisheye",
        # "video.head_center_fisheye",
        # "video.hand_left",
        # "video.hand_right",
        # "video.top_head",
    # ]
    video_keys = [
        "video.hand_left",
        "video.hand_right",
        "video.top_head",
    ]
    state_keys = [
        "state.observation.state",
    ]
    action_keys = [
        "action.action",
    ]

    language_keys = ["annotation.human.task_description"]
    observation_indices = [0]
    action_indices = list(range(16))

    def modality_config(self):
        video_modality = ModalityConfig(
            delta_indices=self.observation_indices,
            modality_keys=self.video_keys,
[... 116 lines cut ...]
         for key in self.apply_to:
-            split_key = key.split(".")
+            split_key = key.split("." ,maxsplit=1)
             assert len(split_key) == 2, "State keys should have two parts: 'modality.key'"
             if key not in self.modality_metadata:
                 modality, state_key = split_key
@@ -401,7 +401,7 @@ class StateActionTransform(InvertibleModalityTransform):

         # Check that all state keys specified in normalization_modes have their statistics in state_statistics
         for key in self.normalization_modes:
-            split_key = key.split(".")
+            split_key = key.split("." ,maxsplit=1)
             assert len(split_key) == 2, "State keys should have two parts: 'modality.key'"
             modality, state_key = split_key
             assert hasattr(dataset_statistics, modality), f"{modality} statistics not found"
@@ -432,7 +432,7 @@ class StateActionTransform(InvertibleModalityTransform):

         # Initialize the normalizers
         for key in self.normalization_modes:
-            modality, state_key = key.split(".")
+            modality, state_key = key.split("." ,maxsplit=1)
             # If the state has a nontrivial rotation, we need to handle it more carefully
             # For absolute rotations, we need to convert them to the target representation and normalize them using min_max mode,
             # since we can infer the bounds by the representation
```

Catchip, 2025-05-31:
I'm trying to evaluate the fine-tuned model in [Genie Sim Benchmark](https://agibot-world.com/sim-evaluation/docs/#/), which is based on issac sim.

youliangtan, 2025-06-30:
This is a sample modality config for the agibot data
```json
{
    "state": {
        "left_arm_joint_position": {
            "original_key": "observation.state",
            "start": 0,
            "end": 7
        },
        "right_arm_joint_position": {
            "original_key": "observation.state",
            "start": 7,
            "end": 14
        },
        "left_effector_position": {
            "original_key": "observation.state",
            "start": 14,
            "end": 15
        },
        "right_effector_position": {
            "original_key": "observation.state",
            "start": 15,
            "end": 16
        },
        "head_position": {
            "original_key": "observation.state",
            "start": 16,
            "end": 18
        },
        "waist_position": {
            "original_key": "observation.state",
            "start": 18,
            "end": 20
        }
    },
    "action": {
        "left_arm_joint_position": {
            "original_key": "action",
            "start": 0,
            "end": 7
        },
        "right_arm_joint_position": {
            "original_key": "action",
            "start": 7,
            "end": 14
        },
        "left_effector_position": {
            "original_key": "action",
            "start": 14,
            "end": 15
        },
        "right_effector_position": {
            "original_key": "action",
            "start": 15,
            "end": 16
        },
        "head_position": {
            "original_key": "action",
            "start": 16,
            "end": 18
        },
        "waist_position": {
            "original_key": "action",
            "start": 18,
            "end": 20
        },
        "robot_velocity": {
            "original_key": "action",
            "start": 20,
            "end": 22
        }
    },
    "video": {
        "top_head_pad_res256_freq10": {
            "original_key": "observation.images.top_head_pad_res256_freq10"
        },
        "hand_left_pad_res256_freq10": {
            "original_key": "observation.images.hand_left_pad_res256_freq10"
        },
        "hand_right_pad_res256_freq10": {
            "original_key": "observation.images.hand_right_pad_res256_freq10"
        }
    },
    "annotation": {
        "language.action_text": {
            "original_key": "annotation.language.action_text"
        },
        "agibot.tasks": {
            "original_key": "annotation.agibot.tasks"
        }
    }
}
```

closed 2025-07-01 by youliangtan