NVIDIA/Isaac-GR00T #191: Fine-Tuning GR00T-N1 on Libero Spatial: Unexpected Impact of Denoise Steps on Performance

Rick0423, 2025-05-31:
First of all, thank you very much for open‐sourcing this incredible project—it has been extremely valuable to the community.

I have been experimenting with fine‐tuning the model on Libero_Spatial dataset, and then benchmarking it in the LIBERO simulation environment. During my tests, I noticed that as I increased the number of denoise steps , the overall accuracy actually decreased rather than improved.

I suspect that this behavior arises because, during fine‐tuning, each forward pass through the model only uses a single DiT pass (i.e., effectively a denoise step of 1). As a result, the distribution of the weights after fine‐tuning is biased toward operating with exactly one denoise iteration. However, at benchmark time, when the policy’s get_action(...) method is called, the model actually applies multiple DiT passes equal to the chosen denoise step count. This mismatch in distributions means that if you choose a denoise step closer to 1 (which matches the fine‐tuning condition), performance is better.

How should I fine‐tune more effectively?  
How should I select the most effective denoising_steps?

Thank you  so much for any suggestions！

LukeLIN-web, 2025-08-10:
libero using single arm, how did you use humanoid robot on it ?

xwjiang2010, 2025-09-16:
> each forward pass through the model only uses a single DiT pass (i.e., effectively a denoise step of 1)

I don't think this statement is true. Each forward pass through the model does not literally correspond to a single DiT pass. Instead, we sample a timestep 𝑡∼𝑈(1,𝑇], add the corresponding level of noise to the input in one shot (using the closed-form formula for the noisy state), and then run the model once to predict the denoised target at that timestep.

Also we have just published the recipes to finetune on Libero datasets. PTAL https://github.com/NVIDIA/Isaac-GR00T/pull/357. 

Feel free to reopen if more issues. Thanks!

sattyamjjain, 2026-09-28:
Your read matches what a train and inference mismatch would predict. One way to confirm it: evaluate the same checkpoint at the denoise step count used in training and at a few higher counts, same seeds and same initial states, and report per task. If the drop is uniform it is the mismatch. If it is one or two tasks, it is something else.