NVIDIA/Isaac-GR00T #200: Challenges with Fine-Tuning UR5 for manipulation tasks with GR00T-N1

jkeshav-bvignesh, 2025-06-06:
I am trying to fine tune the Groot model to do simple pick and place with a UR5 arm. Fine-tuned with 300 episodes (3 tasks, 100 episodes each). The results are not very promising; the robot moves to towards the target location but 7/10 times it will only go to a nearby point (some 5 cm away from the actual point). I have gone through the issues posted in the GitHub page and seems like many other users are also facing similar issues. I am trying to understand more about the model and its implementation on an actual robot, I have the below listed queries:

- Is this just a data problem? Should I collect more data per task? This issue #141 mentions 100 episodes on average.
- If we are using inference in loop, how to understand when the task is completed. The model won't be able to solve all its tasks in one 16-step prediction, right?
- My assumption is that if the predicted values are very close to each other, we can define a stopping criterion. But is that always the case or is there a better way to define the stopping criteria in the finetuning dataset itself?
- In case of multi-step tasks, if we are doing inference in loop, for the consecutive inferences will the model be able to remember the steps it has completed so far?
     eg: go to blue bin and check if it has red pen, if yes pick it else go to black bin and check
- For better results what all robot parameters (joints, TCP pose, joint vel...) should we keep in the states and actions? Currently this is my state and action setup. 

```
"robot_state": [theta1, theta2, theta3, theta4, theta5, theta6, x,y,z,r,p,y,gripper_width]
"robot_action": [theta1, theta2, theta3, theta4, theta5, theta6, x,y,z,r,p,y,gripper_width]

```
where thetas are joint angles, followed by 6D tcp pose and gripper width
- Are there any recommendations for camera placements for better results? My current setup has 2 cameras, gripper view and third person view. 
- From my understanding, the model doesn't accept depth information now. Is that a design choice or is it possible to pass in depth info now?
- Are there any parameters that we can tune during inference to make the results better? This git issue #191 suggests setting the denoising steps closer to 1. But we noticed only marginal differences. Is that the only change?
- For the multi-agent co-ordination shown in this demo, does one model control both robots or is it a "one model instance per robot" design? In that case how does the interaction work? How does the training dataset for this scenario look like. https://www.youtube.com/watch?v=m1CH-mgpdYg&t=86s [1:26 - 1:46]

wangluo2028, 2025-06-08:
I have also finetuned UR5 robot with 100 episodes with only one camera, it performs well, I found the camera pose has greet influence on convergence speed.

wangluo2028, 2025-06-09:
I believe you can finally solve this.

xwjiang2010, 2025-09-09:
@jkeshav-bvignesh Can you also verify that open-loop eval gives your decent result?