NVIDIA/Isaac-GR00T #214: Problems with creating TensorRT engine for new embodiment

NoNoid, 2025-06-15:
Hi, I tried to create a TensorRT engine for our custom dual arm + head camera setup.

The current `deployment_scripts/export_onnx.py` does not handle this case.
So i made the following changes:

- made the embodiment_tag into an argument from and set it to `new_embodiment`
- made the data_config into an argument and set it to a custom config
- pointed `--dataset_path` to a dataset collected using our embodiment
- pointed `--model_path` to a checkpoint trained on that dataset

I get the following error message:

```bash
2025-06-15 18:01:03.670718: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-06-15 18:01:03.695875: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-06-15 18:01:03.695909: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-06-15 18:01:03.696596: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-06-15 18:01:03.700532: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-06-15 18:01:04.251991: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
/opt/conda/envs/gr00t/lib/python3.10/site-packages/albumentations/__init__.py:13: UserWarning: A new version of Albumentations is available: 2.0.8 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
`use_fast` is set to `True` but the image processor class does not have a fast version.  Falling back to the slow version.
Dataset path: data/test_pickplace_dataset
Model path: checkpoints/train_test
ONNX model path: /home/franka-dev/gr00t/gr00t_onnx
Robot data config: humanoid_setup
Robot embodiment tag: new_embodiment
Model not found or avail in the huggingface hub. Loading from local path: checkpoints/train_test
Loading pretrained dual brain from checkpoints/train_test
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: checkpoints/train_test
Tune backbone llm: False
Tune backbone visual: True
Total number of DiT parameters:  550386688
Total number of SelfAttentionTransformer parameters:  201433088
Tune action head projector: True
Tune action head diffusion model: True
Loading checkpoint shards: 100%|██████████| 2/2 [00:00<00:00,  3.86it/s]
Tune backbone llm: False
Tune backbone visual: True
Tune action head projector: True
Tune action head diffusion model: True
Initialized dataset test_pickplace_dataset with new_embodiment
Traceback (most recent call last):
  File "/home/franka-dev/gr00t/deployment_scripts/export_onnx.py", line 481, in <module>
    predicted_action = run_groot_inference(
  File "/home/franka-dev/gr00t/deployment_scripts/export_onnx.py", line 417, in run_groot_inference
    step_data = dataset[0]
  File "/home/franka-dev/gr00t/gr00t/data/dataset.py", line 502, in __getitem__
    return self.transforms(self.get_step_data(trajectory_id, base_index))
  File "/home/franka-dev/gr00t/gr00t/data/dataset.py", line 536, in get_step_data
    data[key] = self.get_data_by_modality(trajectory_id, modality, key, base_index)
  File "/home/franka-dev/gr00t/gr00t/data/dataset.py", line 796, in get_data_by_modality
    return self.get_video(trajectory_id, key, base_index)
  File "/home/franka-dev/gr00t/gr00t/data/dataset.py", line 666, in get_video
    return get_frames_by_timestamps(
  File "/home/franka-dev/gr00t/gr00t/utils/video.py", line 64, in get_frames_by_timestamps
    vr = decord.VideoReader(video_path, **video_backend_kwargs)
  File "/opt/conda/envs/gr00t/lib/python3.10/site-packages/decord/video_reader.py", line 54, in __init__
    self._handle = _CAPI_VideoReaderGetVideoReader(
  File "/opt/conda/envs/gr00t/lib/python3.10/site-packages/decord/_ffi/_ctypes/function.py", line 173, in __call__
    check_call(_LIB.DECORDFuncCall(
  File "/opt/conda/envs/gr00t/lib/python3.10/site-packages/decord/_ffi/base.py", line 78, in check_call
    raise DECORDError(err_str)
decord._ffi.base.DECORDError: [18:01:13] /github/workspace/src/video/video_reader.cc:151: Check failed: st_nb >= 0 (-1128613112 vs. 0) ERROR cannot find video stream with wanted index: -1
ERROR conda.cli.main_run:execute(125): `conda run bash -c python deployment_scripts/export_onnx.py --dataset_path data/test_pickplace_dataset --model_path checkpoints/train_test --data_config_name humanoid_setup --embodiment_tag new_embodiment && bash deployment_scripts/build_engine.sh` failed. (See above for error)
```

kevinsongbing, 2025-06-16:
The error message is from read dataset. Can you fix the error or share the dataset to check the issue? The patch seems to make sense.

youliangtan, 2025-06-16:
@NoNoid I believe this makes sense. Feel free to open a PR for this, we will be happy to merge this in.  Thanks

@kevinsongbing This is because GR00T Nx model uses embodiment tag within the policy,m and data config to load the dataset during runtime. "Vibe looking" this seems make sense.