NVIDIA/Isaac-GR00T #215: NVIDIA GR00T N1.5 FLARE Implementation Inquiry

devedgar, 2025-06-16:
Subject: Request for FLARE Implementation Timeline in GR00T N1.5 Codebase

Dear NVIDIA GR00T Development Team,
I hope this message finds you well. I am writing to inquire about the FLARE (Future Latent Representation Alignment) implementation in the GR00T N1.5 model.
Background
After reviewing the GR00T N1.5 announcement on your research website (https://research.nvidia.com/labs/gear/gr00t-n1_5/), I noticed that FLARE is mentioned as one of the key improvements in N1.5:

"In addition to the flow matching loss used by N1, for N1.5 we add Future LAtent Representation Alignment (see FLARE project). Rather than generatively modeling future frames, FLARE aligns the model with target future embeddings."

The website also shows impressive performance improvements attributed to FLARE integration:

Language following rate: 46.6% → 93.3%
Novel object manipulation: 0% → 15.0% (0-shot)
Significant improvements in various benchmarks

Current Code Analysis
Upon examining the current GR00T N1.5 codebase, I found that while the infrastructure for FLARE is well-prepared, the actual FLARE implementation appears to be inactive:
Evidence of Incomplete FLARE Implementation:

Explicit Comment in Code:
python# flow_matching_action_head.py, line 147
return_all_hidden_states=False,  # NOTE (YL): not using flare now

Missing Future Tokens:

Current sequence: (state_features, action_features)
FLARE requires: (state_features, action_features, future_tokens)

Missing FLARE Loss Components:

No future embedding prediction MLP
No cosine similarity alignment loss
No λ coefficient integration (should be 0.2 according to paper)

Infrastructure Present but Inactive:

return_all_hidden_states flag exists but set to False
DiT architecture supports intermediate layer outputs
Multi-embodiment support through CategorySpecificLinear

Questions

Release Timeline: When can we expect the complete FLARE implementation to be available in the public GR00T N1.5 codebase?
Implementation Status: Is there a specific reason why FLARE is currently disabled in the released code? Is this intentional for the current release?
Documentation: Will there be additional documentation or examples showing how to enable and configure FLARE once it's available?
Compatibility: Will the complete FLARE implementation be backward compatible with existing GR00T N1.5 checkpoints?

Thank you for your time and for making GR00T available to the research community. I look forward to your response.
Best regards,

youliangtan, 2025-06-16:
@devedgar Detailed of Flare paper is in https://arxiv.org/pdf/2505.15659

You will see some reminisces of FLARE in this N1.5 release as we used FLARE during our pretraining stage. Since N1.5 is mainly used for posttraining by the community, we removed FLARE for the sake of cleaner implementation. We would release the full FLARE implementation either as a branch or make it part of N1.5 soon.

FYI, @FrankZheng2022 is the main author for FLARE

devedgar, 2025-06-16:
I'm not sure if I understood this correctly, but does this mean that FLARE itself is not needed during post-training?

youliangtan, 2025-06-17:
Flare is helpful as it is a "future state" prediction, providing more context to the model. But it is optional to use it.