NVIDIA/Isaac-GR00T #225: Robocasa Simulation Environment observation align with GR00T's modality

Jerryisqx, 2025-06-20:
### 🐛 Describe the bug

Hi Everyone,
I'm trying to migrate Robocasa's simulation task on GR00T to verify its performance stated in the report. Now I installed robocasa's envrionment and trying to align robosuite's observation state dict with GR00T's modalidy.json. I'm not familar with the robocasa's embodiment structure and kinetic. Wondering how could I match the state key of single panda gripper with robosuite's env observation key?

Here is state keys that robocasa has:
```
dict_keys(['
robot0_agentview_left_image', 
'robot0_agentview_right_image', 
'robot0_eye_in_hand_image', 
'object', 
'robot0_joint_pos', 
'robot0_joint_pos_cos', 
'robot0_joint_pos_sin', 
'robot0_joint_vel', 
'robot0_eef_pos', 
'robot0_eef_quat', 
'robot0_eef_quat_site', 
'robot0_gripper_qpos', 
'robot0_gripper_qvel', 
'robot0_base_pos', 
'robot0_base_quat', 
'robot0_base_to_eef_pos', 
'robot0_base_to_eef_quat', 
'robot0_base_to_eef_quat_site'])
```
And this is GR00T's modality:
```
"state": {
        "base_position": {
            "start": 0,
            "end": 3
        },
        "base_rotation": {
            "start": 3,
            "end": 7,
            "rotation_type": "quaternion"
        },
        "end_effector_position_absolute": {
            "start": 7,
            "end": 10
        },
        "end_effector_position_relative": {
            "start": 10,
            "end": 13
        },
        "end_effector_rotation_absolute": {
            "start": 13,
            "end": 17,
            "rotation_type": "quaternion"
        },
        "end_effector_rotation_relative": {
            "start": 17,
            "end": 21,
            "rotation_type": "quaternion"
        },
        "gripper_qpos": {
            "start": 21,
            "end": 23
        },
        "gripper_qvel": {
            "start": 23,
            "end": 25
        },
        "joint_position": {
            "start": 25,
            "end": 32
        },
        "joint_position_cos": {
            "start": 32,
            "end": 39
        },
        "joint_position_sin": {
            "start": 39,
            "end": 46
        },
        "joint_velocity": {
            "start": 46,
            "end": 53
        }
```
I know the `robot0_agentview_left_image', 'robot0_agentview_right_image', 'robot0_eye_in_hand_image'` are bascially three cameras. Wondering how can I align for the rest of the robot0 keys. I use [this scripts](https://github.com/Abhiram824/openpi/blob/main/examples/robocasa/convert_robocasa_to_lerobot.py) for the simulation client.

### Versions

Python 3.10.16
absl-py==2.2.2
accelerate==1.2.1
aiosignal==1.3.2
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
astunparse==1.6.3
attrs==25.3.0
av==12.3.0
beartype==0.21.0
bitsandbytes==0.45.5
blessings==1.7
cachetools==5.5.2
certifi==2025.1.31
charset-normalizer==3.4.1
click==8.1.8
cloudpickle==3.1.1
CoLT5-attention==0.11.1
contourpy==1.3.1
cramjam==2.9.1
cycler==0.12.1
decord==0.6.0
diffusers==0.30.2
dm-control==1.0.14
dm-env==1.6
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.16
easydict==1.13
einops==0.8.1
einops-exts==0.0.4
einx==0.3.0
etils==1.12.2
eval_type_backport==0.2.2
evdev==1.9.1
exceptiongroup==1.2.2
Farama-Notifications==0.0.4
-e git+https://github.com/Dao-AILab/fast-hadamard-transform.git@d1a56eee9d502e67faacf61b7b947180d66b32a0#egg=fast_hadamard_transform
fastparquet==2024.11.0
filelock==3.18.0
flash-attn==2.7.1.post4
flatbuffers==25.2.10
fonttools==4.57.0
frozendict==2.4.6
frozenlist==1.5.0
fsspec==2025.3.2
fvcore==0.1.5.post20221221
gast==0.6.0
gitdb==4.0.12
GitPython==3.1.44
glfw==2.8.0
google-auth==2.38.0
google-auth-oauthlib==1.2.1
google-pasta==0.2.0
-e git+https://github.com/Jerryisqx/Isaac-GR00T.git@1452df44e4fa20a76c8bce90910cc8ae6a300d50#egg=gr00t
grpcio==1.71.0
gym-aloha==0.1.1
gymnasium==1.0.0
h5py==3.12.1
huggingface-hub==0.29.1
hydra-core==1.3.2
hyper-connections==0.2.1
idna==3.10
imageio==2.34.2
imageio-ffmpeg==0.6.0
importlib_metadata==8.6.1
importlib_resources==6.5.2
iniconfig==2.1.0
iopath==0.1.10
Jinja2==3.1.6
joblib==1.3.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
keras==2.15.0
kiwisolver==1.4.8
kornia==0.7.4
kornia_rs==0.1.8
labmaze==1.0.6
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
local-attention==1.11.1
loguru==0.7.3
lxml==5.3.2
Markdown==3.7
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
mdurl==0.1.2
mink==0.0.7
ml-dtypes==0.2.0
mpmath==1.3.0
msgpack==1.1.0
mujoco==3.3.1
networkx==3.4.2
ninja==1.11.1.4
numba==0.61.2
numpy==1.26.4
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-cusparselt-cu12==0.6.2
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.2.2
omegaconf==2.3.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
-e git+https://github.com/Jerryisqx/Isaac-GR00T.git@1452df44e4fa20a76c8bce90910cc8ae6a300d50#egg=openpi_client&subdirectory=sim/openpi-client
opt_einsum==3.4.0
packaging==24.2
pandas==2.2.3
peft==0.14.0
pettingzoo==1.24.3
pillow==11.1.0
pipablepytorch3d==0.7.6
platformdirs==4.3.7
pluggy==1.5.0
portalocker==3.1.1
protobuf==3.20.3
psutil==7.0.0
pyarrow==14.0.1
pyasn1==0.6.1
pyasn1_modules==0.4.2
pyav==14.2.1
pydantic==2.10.6
pydantic_core==2.27.2
Pygments==2.19.1
pynput==1.8.1
PyOpenGL==3.1.9
pyparsing==3.2.3
pytest==8.3.5
python-dateutil==2.9.0.post0
python-xlib==0.33
pytz==2025.2
PyYAML==6.0.2
pyzmq==26.4.0
qpsolvers==4.5.1
quadprog==0.1.13
ray==2.40.0
referencing==0.36.2
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
rich==14.0.0
# Editable install with no version control (robosuite==1.5.1)
-e /scratch_net/biwidl313/robosuite
rpds-py==0.24.0
rsa==4.9
safetensors==0.5.3
scikit-image==0.25.2
scikit-learn==1.5.2
scipy==1.15.2
sentry-sdk==2.25.1
setproctitle==1.3.5
shtab==1.7.1
six==1.17.0
smmap==5.0.2
svgwrite==1.4.3
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==2.5.0
threadpoolctl==3.6.0
tianshou==0.5.1
tifffile==2025.3.30
timm==1.0.14
tokenizers==0.20.3
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tqdm==4.67.1
transformers==4.45.2
Tree==0.2.4
triton==3.1.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
urllib3==2.3.0
vector-quantize-pytorch==1.22.17
wandb==0.18.0
websockets==15.0.1
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zetascale==2.8.6
zipp==3.21.0
zmq==0.0.0

LukeLIN-web, 2025-06-25:
same problem, did you solve it?

Jerryisqx, 2025-06-29:
Now I had an issue, I finetuned robocasa tasks (e.g. close drawer) for 64 per gpu batch on 2 devices currently. When I used the 15k steps checkpoint, here is what I observed:
There are some tasks actually finished the job, but the simulator still reported as 'failed'. (For closer drawer, actually 5 trials are success in total of 10) I wonder is it an issue with the simulation configuration?