NVIDIA/Isaac-GR00T #237: Tensorrt how to fit two camera?

zmf2022, 2025-07-02:
I encounter an error as follows:
IExecutionContext::getTensorShape: Error Code 7: Internal Error (Dimensions with name batch_size must be equal. Condition '==' violated: 1 != 2. Instruction: CHECK_EQUAL 1 2.)

wytalfred, 2025-07-03:
I got the same problem. waiting for solutions...

zmf2022, 2025-07-04:
1. export_onnx.py p234 modify as  "vit_embeds": {0: "batch_size", 1: "vit_seq_len"},
2. trt_model_forward.py p45 add vit_embeds = vit_embeds.view(1, -1, vit_embeds.shape[-1])  # [2, 256, 1152] -> [1, 512, 1152]
3. build_engine.sh modify as  MAX_LEN=600   
# VLM-LLM
echo "------------Building VLM-LLM--------------------"
trtexec --verbose --stronglyTyped --separateProfileRun --noDataTransfers --onnx=gr00t_onnx/eagle2/llm.onnx  --saveEngine=gr00t_engine/llm.engine --minShapes=input_ids:1x${MIN_LEN},vit_embeds:1x256x1152,attention_mask:1x${MIN_LEN} --optShapes=input_ids:1x${OPT_LEN},vit_embeds:1x512x1152,attention_mask:1x${OPT_LEN} --maxShapes=input_ids:8x${MAX_LEN},vit_embeds:8x512x1152,attention_mask:8x${MAX_LEN} > gr00t_engine/llm.log 2>&1