NVIDIA/Isaac-GR00T #238: Inquiry on Embodiment Tag Used for RoboCasa Data

qqz03, 2025-07-02:
Hi, thank you very much for the great work and detailed paper!

I noticed that the paper mentions using 3,000 samples from the **RoboCasa** dataset based on the **Panda robot arm**, rather than the **GR1 humanoid robot**.
Could you kindly clarify which **Embodiment Tag** was used during the **post-training** and/or **evaluation** phases when using **RoboCasa** data?

Understanding this would greatly help us ensure consistency when reproducing your results.

Many thanks in advance for your time and help!

Jerryisqx, 2025-07-02:
I think for 24 robocasa  kitchen tasks, they use single_panda_gripper?

youliangtan, 2025-07-16:
For the 24 kitchen tasks, I'm using
```
--data_config fourier_gr1_arms_waist  --embodiment_tag gr1 
```

closed 2025-07-16 by youliangtan