NVIDIA/Isaac-GR00T #245: Whether fintune the diffusion head when finetuning on new embodiment

jrryzh, 2025-07-07:
### 📚 The doc issue

I have a question about whether to finetune the diffusion head during finetuning, it seems that in the "get deeper understanding " md, it says "By default, the diffusion model is not tuned"; but in the script finetune_gr00t_policy.py, the parameter is set to true by default. So it confuses me whether to tune the weights of diffusion head.

PS: in practice i tried tune the diffusion head when tuning for a new embodiment, it takes a lot of tuning steps to perform well and i wonder whether this has to do with it. Thanks for answering!

youliangtan, 2025-07-09:
I would recommend finetune it if you have sufficient GPU ram. The doc is not updated.

jrryzh, 2025-07-09:
Thanks, I will try some more experients and see if it gets better

closed 2025-07-16 by youliangtan