NVIDIA/Isaac-GR00T #247: Meet DDP error when opening visual tunning for GR00T1.5

Jerryisqx, 2025-07-08:
### 🐛 Describe the bug

Hi, when I use the GR00TN1.5 finetuning on LIBERO dataset, I met a DDP error when opening the tune_visual option. It returns:
```

  0%|          | 1/25000 [00:09<65:42:33,  9.46s/it]Traceback (most recent call last):
  File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/scripts/gr00t_finetune.py", line 280, in <module>
    main(config)
  File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/scripts/gr00t_finetune.py", line 249, in main
    experiment.train()
  File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/runner.py", line 171, in train
    self.trainer.train(resume_from_checkpoint=self.resume_from_checkpoint)
  File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/trainer.py", line 153, in train
    return super().train(resume_from_checkpoint, trial, ignore_keys_for_eval, **kwargs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 2245, in train
    return inner_training_loop(
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 2560, in _inner_training_loop
    tr_loss_step = self.training_step(model, inputs, num_items_in_batch)
[rank1]: Traceback (most recent call last):
[rank1]:   File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/scripts/gr00t_finetune.py", line 280, in <module>
[rank1]:     main(config)
[rank1]:   File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/scripts/gr00t_finetune.py", line 249, in main
[rank1]:     experiment.train()
[rank1]:   File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/runner.py", line 171, in train
[rank1]:     self.trainer.train(resume_from_checkpoint=self.resume_from_checkpoint)
[rank1]:   File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/trainer.py", line 153, in train
[rank1]:     return super().train(resume_from_checkpoint, trial, ignore_keys_for_eval, **kwargs)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 2245, in train
[rank1]:     return inner_training_loop(
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 2560, in _inner_training_loop
[rank1]:     tr_loss_step = self.training_step(model, inputs, num_items_in_batch)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 3736, in training_step
[rank1]:     loss = self.compute_loss(model, inputs, num_items_in_batch=num_items_in_batch)
[rank1]:   File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/trainer.py", line 75, in compute_loss
[rank1]:     outputs = model(inputs)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1736, in _wrapped_call_impl
[rank1]:     return self._call_impl(*args, **kwargs)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1747, in _call_impl
[rank1]:     return forward_call(*args, **kwargs)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/parallel/distributed.py", line 1639, in forward
[rank1]:     inputs, kwargs = self._pre_forward(*inputs, **kwargs)
[rank1]:   File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/parallel/distributed.py", line 1528, in _pre_forward
[rank1]:     if torch.is_grad_enabled() and self.reducer._rebuild_buckets():
[rank1]: RuntimeError: Expected to have finished reduction in the prior iteration before starting a new one. This error indicates that your module has parameters that were not used in producing loss. You can enable unused parameter detection by passing the keyword argument `find_unused_parameters=True` to `torch.nn.parallel.DistributedDataParallel`, and by 
[rank1]: making sure all `forward` function outputs participate in calculating loss. 
[rank1]: If you already have done the above, then the distributed data parallel module wasn't able to locate the output tensors in the return value of your module's `forward` function. Please include the loss function and the structure of the return value of `forward` of your module when reporting this issue (e.g. list, dict, iterable).
[rank1]: Parameter indices which did not receive grad for rank 1: 437 438 439 440 441 442 443 444 445 446 447
[rank1]:  In addition, you can set the environment variable TORCH_DISTRIBUTED_DEBUG to either INFO or DETAIL to print out information about which particular parameters did not receive gradient on this rank as part of this error
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/transformers/trainer.py", line 3736, in training_step
    loss = self.compute_loss(model, inputs, num_items_in_batch=num_items_in_batch)
  File "/srv/beegfs02/scratch/qingxuan_project/data/GR00T/Isaac-GR00T-N1-5/gr00t/experiment/trainer.py", line 75, in compute_loss
    outputs = model(inputs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1736, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1747, in _call_impl
    return forward_call(*args, **kwargs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/parallel/distributed.py", line 1639, in forward
    inputs, kwargs = self._pre_forward(*inputs, **kwargs)
  File "/scratch_net/biwidl313/chenqing/conda_envs/robocasa-dev/lib/python3.10/site-packages/torch/nn/parallel/distributed.py", line 1528, in _pre_forward
    if torch.is_grad_enabled() and self.reducer._rebuild_buckets():
RuntimeError: Expected to have finished reduction in the prior iteration before starting a new one. This error indicates that your module has parameters that were not used in producing loss. You can enable unused parameter detection by passing the keyword argument `find_unused_parameters=True` to `torch.nn.parallel.DistributedDataParallel`, and by 
making sure all `forward` function outputs participate in calculating loss. 
If you already have done the above, then the distributed data parallel module wasn't able to locate the output tensors in the return value of your module's `forward` function. Please include the loss function and the structure of the return value of `forward` of your module when reporting this issue (e.g. list, dict, iterable).
Parameter indices which did not receive grad for rank 0: 437 438 439 440 441 442 443 444 445 446 447
 In addition, you can set the environment variable TORCH_DISTRIBUTED_DEBUG to either INFO or DETAIL to print out information about which particular parameters did not receive gradient on this rank as part of this error
```
I tried to set the `ddp_find_unused_parameters` in training config to True but the problem is exists.

### Versions

[... 155 lines cut ...]
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==3.1.0
tianshou==0.5.1
tifffile==2025.5.10
timm==1.0.14
tokenizers==0.21.2
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tqdm==4.67.1
transformers==4.51.3
triton==3.1.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
urllib3==2.5.0
wandb==0.18.0
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp==3.23.0

a1037441813, 2025-07-14:
The same problem occurred.

you can freeze  head block of visual model .the problem will can be solved.

#####self.eagle_model.vision_model.vision_model.head.requires_grad_(False)

or  
you can open the LLM model,,,,,the problem also can be solved...

closed 2025-07-16 by youliangtan

BlackTea-c, 2025-07-31:
i tried to finetune llm+diffusion model but still reported this error....

zjh908491372, 2025-08-20:
Resolved
Encounter the same problem, but the indices are different.
Use TORCH_DISTRIBUTED_DEBUG=INFO to check which parameter does not participate in the forward, and change the corresponding parameter to model.requires_grad_ (False) in the corresponding set_trainable_parameters function.