NVIDIA/Isaac-GR00T #251: Very poor training performance with GR00T on the PhysicalAI-Robotics-GR00T-Teleop-Sim dataset for robocasa-gr1-tabletop-tasks

4133223, 2025-07-09:
We are using GR00T for training on the dataset mentioned in robocasa-gr1-tabletop-tasks, which is available at https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-GR00T-Teleop-Sim. The training parameters are as follows:
[image]
We have tried loading either a single dataset or multiple datasets for training, but the model performance is very poor. When loading multiple datasets for training, the test success rate of most tasks is 0, and the maximum for a small number of tasks is only around 30%.

34328, 2025-07-09:
I meet the same problems, i just SFT in ''gr1_unified.PnPCanToDrawerClose'' task dataset，The success rate is also very low，10%？

yeshas1994, 2025-07-11:
I have also tested finetuning PnPCanToDrawer (all 1000 trajectories) for 30,000 - 60,000 steps and consistently get around a 35-45% success rate.

4133223, 2025-07-11:
@yeshas1994 My conclusion aligns with yours. I also fine-tuned the latest model specifically for the single task of PnPCanToDrawer, and the success rate hovers around 40%. Of course, RoboCasa's success criteria can sometimes be inconsistent—it might misclassify a successfully completed task as a failure. When I fine-tuned the model on multiple tasks simultaneously, I noticed a clear trend: the more tasks I included, the worse the performance on individual tasks became. I'm curious—were the results in the paper achieved by fine-tuning on all tasks concurrently?

youliangtan, 2025-07-11:
Similar issue: https://github.com/NVIDIA/Isaac-GR00T/issues/249

With my small scale single task env. Im able to get around 50% for Cup to Drawer Task.
```bash
python scripts/gr00t_finetune.py \
--dataset-path datasets/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/gr1_arms_waist.CupToDrawer \
--num-gpus 2 --batch-size 25  --output-dir ~/checkpoints/24dc-cup-drawer-09jul-tune-visual-2gpu  \
--data-config fourier_gr1_arms_waist --max-steps 16000 \
--embodiment_tag gr1 --save-steps 2000 --tune-visual
```

```bash
python3 scripts/simulation_service.py --client     \
--env_name gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env  \
   --video_dir ./videos4     --max_episode_steps 720     --n_envs 5     --n_episodes 10

....
Collecting 10 episodes took 99.85 seconds
Results for gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env:
Success rate: 0.50
```

Note i use `--tune-visual` in this case.

https://github.com/user-attachments/assets/a13a1eec-5fe6-4649-8d83-46df2ecc1dbe

Checking out others when i have more time soon

4133223, 2025-07-11:
@youliangtan Thank you for your reply. Currently, I am also experimenting with the --tune-visual parameter. However, this has led to some issues with multi-GPU training, so I am now using single-GPU training and hoping for good results.

youliangtan, 2025-07-11:
@4133223 i have a temp fix for this here: https://github.com/NVIDIA/Isaac-GR00T/pull/257

youliangtan, 2025-07-11:
@4133223 i have a temp fix for multigpu visual tuning here: https://github.com/NVIDIA/Isaac-GR00T/pull/257

youliangtan, 2025-07-11:
Also noted that it is important to run the with more number of episodes during rollout eval `n_episodes`. I tried with just 10 initially and the variance is quite high, so run it with 30 now.

```bash
python scripts/gr00t_finetune.py --dataset-path /home/youliang/gear/clean_repo/Isaac-GR00T/datasets/PhysicalAI-Robotics-GR00T-X-Embodiment-Sim/gr1_arms_waist.CuttingboardToPan --num-gpus 2 --batch-size 25  --output-dir ~/checkpoints/24dc-board-pan-10jul-tune-visual-2gpu  --data-config fourier_gr1_arms_waist --max-steps 10000 --embodiment_tag gr1 --save-steps 2000 --tune-visual

# SR = 63% 
python3 scripts/simulation_service.py --client   \
  --env_name gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env \
  --video_dir ./videos-b2p \
  --max_episode_steps 720 \
  --n_envs 5 \
  --n_episodes 30
```

4133223, 2025-07-15:
I have tried to load 24 tasks simultaneously for fine-tuning, with the tune_visual parameter set to false. However, based on my fine-tuning results, the model is unable to complete tasks that require coordinated operation of both hands. In some of the simulation results, there were indeed cases where the model failed to correctly recognize the position of the cup. I suspect that this might be because the visual information of the model has not been fully utilized.

Nevertheless, in the fine-tuning for a single task (such as PnPCupToDrawerClose), the model can sometimes even correct its failures. For example, if it first knocks the cup over, it can then perform the action again and pick up the cup.( with the tune_visual parameter set to false)

4133223, 2025-07-15:
@icyreq When performing fine-tuning for a single task, the parameters I used are as follows: steps are set to 30,000, tune-visual is false, batch size is 128, and 6 GPUs are employed.

icyreq, 2025-07-15:
@4133223 
Thank you for your suggestions. Your training volume is many times larger than mine. It seems the issue is likely not overfitting, but rather that the model is undertrained and hasn't yet learned to form the complex association between visual information and robot actions. I will try again.

4133223, 2025-07-15:
However, I haven't tested the impact of the initial position of the moving cup on the test results either.

4133223, 2025-07-15:
However, I haven't conducted any tests on how the initial position of the cup affects the test results either.

youliangtan, 2025-07-15:
I'm getting much decent success rate after https://github.com/NVIDIA/Isaac-GR00T/pull/264. Need not `tune-visual`. Try it and let me know.

youliangtan, 2025-07-21:
Since im not getting more issues response. i will close this issue ticket for now. Any new issue feel free to open a new ticket 😃 

-------------------

Drop in some details here: 

I'm using first 1000 trajs of each tasks for training. The training script is below on a single node h100.

```bash
python scripts/gr00t_finetune.py \
  --dataset-path "${ALL_DATASET_PATHS[@]}" \
  --num-gpus 8 --batch-size 120 --learning_rate 3e-5 \
  --output-dir /mnt/youliangt/datasets/checkpoints/24dc-all-30k-11jul \
  --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
  --max-steps 30000 --save-steps 2000
```

I'm getting on avg 45% for all tasks. More hyperparam tuning or enabling flare training will result in better perf. But at leastt now the results or close to what mentioned in the whitepaper.

closed 2025-07-21 by youliangtan

HeanyGao, 2025-08-04:
> Drop in some details here:
Hello, I used the same parameters to perform fine-tuning on 24 datasets for multi-task learning, and selected a 30k checkpoint. However, the average success rate of the tasks was less than 10%. Could you please explain how you selected the optimal checkpoint?

LukeLIN-web, 2025-08-08:
> achieved by fine-tuning on all tasks concurrently?

I think they fine tune on each task separately

LukeLIN-web, 2025-08-08:
> Since im not getting more issues response. i will close this issue ticket for now. Any new issue feel free to open a new ticket 😃
> 
> Drop in some details here:
> 
> I'm using first 1000 trajs of each tasks for training. The training script is below on a single node h100.
> 
> python scripts/gr00t_finetune.py \
>   --dataset-path "${ALL_DATASET_PATHS[@]}" \
>   --num-gpus 8 --batch-size 120 --learning_rate 3e-5 \
>   --output-dir /mnt/youliangt/datasets/checkpoints/24dc-all-30k-11jul \
>   --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
>   --max-steps 30000 --save-steps 2000
> I'm getting on avg 45% for all tasks. More hyperparam tuning or enabling flare training will result in better perf. But at leastt now the results or close to what mentioned in the whitepaper.

Did you train all  task dataset together? Or train it one by one?

LukeLIN-web, 2025-08-09:
> Since im not getting more issues response. i will close this issue ticket for now. Any new issue feel free to open a new ticket 😃
>
> Drop in some details here:
>
> I'm using first 1000 trajs of each [...]

I run this train only on `datasets/LeRobot/gr1_unified.PnPCupToDrawerClose` and eval 
Results for gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env:
Success rate: 0.30 .  n_episodes = 10 .