NVIDIA/Isaac-GR00T #260: Different --n_envs settings lead to a striking sim-eval performance (SR) gap

qqz03, 2025-07-13:
I'm observing an interesting phenomenon when evaluating the fine-tuned checkpoints from `gr1_arms_waist.CanToDrawer` on the single-task `"gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env"`. 

### Setup Details:
- **Fine-tuning** (using all 10k trajectories from HF dataset gr1_arms_waist.CanToDrawer):
  ```bash
  TUNE_VISUAL=true
  NUM_GPUS=1
  MAX_STEPS=7000
  SAVE_STEPS=1000
  BATCH_SIZE=64
  DATA_CONFIG="fourier_gr1_arms_waist"
  EMBODIMENT_TAG="gr1"
  ```
- **Evaluation**:
  ```bash
  python3 scripts/simulation_service.py --client \
              --env_name "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env" \
              --video_dir "$VIDEO_DIR" \
              --max_episode_steps 720 \
              --n_envs ${N_ENV} \
              --n_episodes 50
  ```

### Key Observation:
1. The success rate varies dramatically with different `--n_envs` values, despite:
   - Using identical model checkpoints
   - Having sufficient GPU memory in all cases
   - Keeping all other parameters constant
2. **Video Verification**: I've manually checked the saved videos and confirmed:
   - No fake success (actual failure marked as success)
   - No fake failure (actual success marked as failure)

### Results:

| Checkpoint   | n_envs | Time (s) | Success Rate |
|--------------|--------|----------|--------------|
| checkpoint-1000 | 5      | 523.86   | **0.52**         |
| checkpoint-1000 | 50     | 727.07   | **0.00**         |
| checkpoint-2000 | 5      | 522.00   | 0.30         |
| checkpoint-2000 | 50     | 703.43   | 0.02         |
| checkpoint-3000 | 5      | 393.51   | 0.14         |
| checkpoint-3000 | 50     | 697.29   | 0.00         |

### Questions:
1. Is this level of sensitivity to `--n_envs` expected behavior?
2. Could there be any hidden interactions between parallel environments affecting the policy?
3. Would you recommend specific `--n_envs` ranges for reliable evaluation?

Thanks for your insights! The performance gap between **n_envs=5 (52% success)** and **n_envs=50 (0% success)** at checkpoint-1000 is particularly striking.

@youliangtan 

#249 #251 #253

youliangtan, 2025-07-14:
Tbh im not too sure, need to debug it further. `n_envs` will run the env parallely via different cpu processes i think. From rough guess, i think having 50 `n_envs` might got blocked by the cpu resources, resulted in a timeout hence task incompletion.

If things still presists, can try to open another issue ticket in https://github.com/robocasa/robocasa-gr1-tabletop-tasks, which links to this issue.

Also, a later checkpoint might result in overfiting in the policy, can try if use default `tune-visual=false`, which makes it overfit.

qqz03, 2025-07-14:
[https://github.com/robocasa/robocasa-gr1-tabletop-tasks/issues/4]

**`--episodes=100`**

### More Results:
| Model           | n_envs | Time (s) | Success Rate | GPU MEM Usage |
|-----------------|--------|----------|--------------|---------------|
| checkpoint-1000 | 5      | 1002.45  | **0.58**         | 15%           |
| checkpoint-1000 | 10     | 654.77   | **0.41**         | 21%           |
| checkpoint-1000 | 20     | 600.96   | **0.22**         | 33%           |
| checkpoint-1000 | 50     | 862.84   | **0.02**         | 78%           |

4133223, 2025-07-15:
@qqz03 I wonder, after the fine-tuning is completed, what value did your loss reach?

qqz03, 2025-07-17:
Hi @4133223 @Jerryisqx 

I fine-tuned the model on the `gr1_arms_waist.CanToDrawer` dataset from the `PhysicalAI-Robotics-GR00T-X-Embodiment-Sim` repo, using the full set of 10k+ trajectories.

At **step 1000**, the loss was **0.0151**, and the **max_steps** was **1700**, where the loss was **0.0152**.

Here’s the training log:
```json
{
  "epoch": 0.0338895201642645,
  "step": 1700,
  "total_flos": 0.0,
  "train_loss": 0.03195767214193063,
  "train_runtime": 3894.0888,
  "train_samples_per_second": 27.94,
  "train_steps_per_second": 0.437
}

LukeLIN-web, 2025-08-10:
I think this is because more diverse environments require higher generalization ability.