NVIDIA/Isaac-GR00T #270: Finetuned and inference on a new embodiment

kc645, 2025-07-17:
Hello,
thank you for all your hard work on Gr00t!
I am trying to finetune a task on my robot, which is a dual 7 DoF manipulator.

I have created a 201 datasets of 20 seconds per episode. I trained on 200 datasets and left one for inference on test dataset.

https://github.com/user-attachments/assets/23faa1b7-e8a0-4ebb-9fda-2aa5405c8489

After finish training, I tested on a new episode and the loss was 0.0003 (on both pytorch model and tensorrt engine). I am using Nvidia Jetson AGX Orin 64 Gb to test. 

[image]

However, when I tried to inference on my actual hardware, the robot does not inference very well. 
This is the part of the code i wrote to inference.

def eval_bc(self, config):
    """
    Evaluates the behavior cloning model using the specified configuration and checkpoint.

    Args:
        config (dict): Configuration parameters for evaluation.
    """
    self.max_timesteps = self.args.steps
    camera_names = ["cam_right_ee", "cam_left_ee"]

    # Only start cameras if they haven't been started yet
    if not hasattr(self.env, 'camera_names') or self.env.camera_names != camera_names:
        self.node.get_logger().info("Starting cameras...")
        self.env.start_cam(camera_names)
    else:
        self.node.get_logger().info("Cameras already started, skipping camera initialization")

    query_frequency = 1
    self.num_queries = self.args.action_horizon
    TIME_MULTI = 0.1  # TODO: make this cli arg
    self.max_timesteps = int(self.max_timesteps * TIME_MULTI)

    self.t = 0

    self.status_bit = 0.0
    obs = self.env.reset()

    save_input_data_path = f"test_going_in/input_data.pkl"
    self.node.get_logger().info(f"self.args: {self.args}")
    
    # Initialize data collection arrays for plotting
    state_data_across_time = []
    pred_action_data_across_time = []
    step_count = 0
    while step_count < 1000:
        with self.action_mutex:
            if self.status_bit > 1.5:
                break
        
        current_obs = self.env.get_observation()
        start_time = time.time()
        action_chunk = self.policy.get_action(current_obs)
        end_time = time.time()
        inference_time = end_time - start_time
        hz = 1.0 / inference_time
        print(f"Inference time: {inference_time:.4f} seconds ({hz:.1f} Hz)")
        state_arrays = []
        for key in self.args.modality_keys:
            state_val = current_obs[f"state.{key}"]
            
            # Ensure the value is at least 1-dimensional
            if state_val.ndim == 0:  # scalar
                state_arrays.append(np.array([state_val]))
            elif state_val.ndim > 1:  # multi-dimensional
                state_arrays.append(state_val.flatten())
            else:  # already 1-dimensional
                state_arrays.append(state_val)
        state_concat = np.concatenate(state_arrays, axis=0)
        state_data_across_time.append(state_concat)
        self.inference_point = 16
        for i in range(self.inference_point):
            step_count += 1
            current_action = {}
            for key, value in action_chunk.items():
                current_action[key] = value[i]
            current_obs = self.env.get_observation()
            action_arrays = []
            for key in self.args.modality_keys:
                action_val = current_action[f"action.{key}"]
                # Ensure the value is at least 1-dimensional
                if action_val.ndim == 0:  # scalar
                    action_arrays.append(np.array([action_val]))
                elif action_val.ndim > 1:  # multi-dimensional
                    action_arrays.append(action_val.flatten())
                else:  # already 1-dimensional
                    action_arrays.append(action_val)
            action_concat = np.concatenate(action_arrays, axis=0)
            pred_action_data_across_time.append(action_concat)
            env_start_time = time.time()
            self.env.step(current_action)
            env_end_time = time.time()
            env_time = env_end_time - env_start_time
            env_hz = 1.0 / env_time

        
        current_obs = self.env.get_observation()
        self.node.get_logger().info(f"-------------------------------------------------")

        self.env.input_data_across_time.append(obs)
        step_count += 1

    # Create directory if it doesn't exist
    save_dir = os.path.dirname(save_input_data_path)
    if save_dir:  # Only create directory if there's a directory component
        os.makedirs(save_dir, exist_ok=True)
    with open(save_input_data_path, 'wb') as f:
        pickle.dump({
            'input_data_per_step': self.env.input_data_across_time
        }, f)
    print(f"Input data saved to: {save_input_data_path}")
    print(f"Saved {len(self.env.input_data_across_time)} steps of input data")

    print("Generating inference plots...")
    self.plot_inference_results(
        state_data=state_data_across_time,
        pred_action_data=pred_action_data_across_time,
        modality_keys=self.args.modality_keys,
        action_horizon=self.inference_point,
    )

    return

I would appreciate any tips

kc645, 2025-07-17:
I tried plotting the actions when running inference on the real hardware

[image]

youliangtan, 2025-07-18:
Something seems wrong. just briefly looking at the issue, some comments:
1. The video you uploaded isnt working
2. the openloop plot is okay, but not great. i think this might need to train longer or larger batchsize.
3. the inference plot is very sus, huge fluctuations. Maybe the input is incorrect?

kc645, 2025-07-18:
https://github.com/user-attachments/assets/d2d8e1fb-5b72-469c-9c91-bf7a5afbe7ee

1. I have attached the video again
2. I have trained with batch size of 32 with lr of 1e-4 and I am only finetuning projector and diffusion model. Do you recommend otherwise?
3. I am still looking into the inference plot because the fluctuation is also weird. I have also attached the very first inference of both open loop and inference. From the open loop it seems as if the action horizon is smooth but the inference the action loop is not as smooth. 

[image]

[image]

I would assume at least the first inference should be similar.

OpenJarvisAI, 2025-07-24:
Hit simillar issues as yours.

to me, got 150 episodes and 2 tasks total, trained about 10000 steps, not working at all. (loss converged into 0.0003)

kc645, 2025-07-24:
did you try something that was successful? or trying to do something?

OpenJarvisAI, 2025-08-01:
The model is not totally not working, here is my observation:

1. I tried training in robotwin, it works but is janky, shaking, even though the task could succeed (sr not high);
2. Using a real arm, it's extremely janky and shaking, and can't grab things.

Now, we have no idea what's going wrong.

From your graph attach compare inference actions and state, seems they are not aligned, the current action should be the next state, but seems not