NVIDIA/Isaac-GR00T #272: Robot deployment GR00T inference strategy

edisimo, 2025-07-18:
Hello! I am trying to deploy GR00T finetuned on a new embodiment and I was wondering what the best practice / optimal way to infer the actions is.

- Am I supposed to plan once, execute the whole action horizon and then plan again? 
- Or is it better to run the inference somewhere in the middle of the horizon while executing the actions?
- Or perhaps just to take the first step and replan immediately afterwards? 
- ......?

I am currently trying to execute the whole action horizon and then replan, however I am having some trouble like the manipulator returning a few steps back everytime I recieve the new action horizon. I’m curious as to what the recommended way to handle horizon inference is to achieve smooth, continuous motion.

Thank you for your help and for all the great work on GR00T!

kc645, 2025-07-18:
Hi, I am running into a similar problem as you. I am trying all three of your strategies(?) but not great results yet.
Mine seems to also move back.

Just curious, how many datasets did you finetune on a new embodiment?
I collected 200 datasets of 20 seconds tasks.

anarnuri, 2025-07-18:
Hi, I had the same question before. I tried fine-tuning on Isaac Lab generated dataset of pick and place for SO100 arm and found out that going through all 16 predicted actions worked best. If I would only take the next prediction and do the inference a lot more often it would decrease the performance and sometimes the robot would simply get stuck at the same point. 

Also increasing the denoising steps during inference from 4 (this is what they set it at) to say 100, 250 steps did not help and would take a lot more time.