NVIDIA/Isaac-GR00T #276: availability of finetuned model for inference test

jcl2023, 2025-07-19:
### 🚀 The feature, motivation and pitch

My gpu is old, 2080ti and is not able to do finetuning a model.
Is there a finetuned model available so I can do evaluation?

Thanks

jcl2023, 2025-07-22:
My robot is Lerobot SO-101. Is there a finetuned model available for this robot?

Thanks,

jcl2023, 2025-08-19:
Thanks for an update,

jcl2023, 2025-08-20:
@anarnuri 

    I like to try it if you provide me the model and an instruction of how to do inference with your finetuned model.
My so-arm101 has not arrived yet. Is it possible to do inference with isaac-sim?

Thanks,

anarnuri, 2025-08-21:
@jcl2023 anurizada/pick_so100_10000_steps and anurizada/pick_dataset

jcl2023, 2025-08-27:
@anarnuri 

    Thanks for an update。 When I get a new GPU card which supports flash-attn, I will try your finetuned model