NVIDIA/Isaac-GR00T #277: Help: Problem Encountered While Post-Training Isaac GR00T N1.5 for LeRobot SO-101 Arm per the Guide

uclick-ljw, 2025-07-22:
Thank you to the authors of this document.
https://huggingface.co/blog/nvidia/gr00t-n1-5-so101-tuning?ncid=so-disc-442150-vt48

I followed this document to perform post-training on so101.

After completing post-training, I followed steps 3 and 4,
but so101 is not functioning properly.

If you look at the comments in this document, the phenomenon shown in the attached photo occurs.
I experienced the same phenomenon with the actual so101.
I didn't attach it here because it's not a video I filmed.

[image]

Have you ever experienced a similar issue and resolved it?

I can't identify where the problem occurred,
so I am currently trying to change the learning parameters or do post-training with a different dataset.

If anyone has experience solving a similar problem, please respond.

Once again, I want to express my gratitude to those who wrote the guide.

uclick-ljw, 2025-07-23:
https://cdn-uploads.huggingface.co/production/uploads/66ebc3fdcfb61a283797c771/ppNrMUS5FR6Q2ck7c1Pnu.webm

HomerIsAFool, 2025-07-25:
Do you use both fixed camera and hand-eye camera as observation input?

uclick-ljw, 2025-07-27:
@HomerIsAFool 
Yes, I'm using wrist and front camera like this

[image]

If I explain the method I used, it is as shown in the link below
https://hyper-quark-559.notion.site/SO-101-GR00T-23d59bee00b9806bbdcae7bcbf8ee559?source=copy_link

I proceeded almost exactly according to the [guide](https://huggingface.co/blog/nvidia/gr00t-n1-5-so101-tuning?ncid=so-disc-442150-vt48)

Finally,
The system I am using is as follows

CPU: Intel® Xeon® W5-3435X
Memory: 512
GPU: A5500 X 2
OS: Ubuntu 22.04.05, desktop

HomerIsAFool, 2025-07-28:
@uclick-ljw  i also tried the gr00t n1.5 for so101 and the result is rather bad either, i think this is because the gap between official dataset and our experiment scene. The model’s generalization ability isn’t that strong as they mentioned.

Maybe we can collect some data ourselves and retry.

By the way, i tried the gr00t-n1 using unitree G1 stack blocking dataset, and the result is bad as well.

uclick-ljw, 2025-07-28:
@HomerIsAFool 
Thank you,
I will generate training data in the current environment and try training.

I hope the reason it didn't work well was due to the training data.
If the cause is other issues like different parameter values or packages, it will be difficult to find.

youliangtan, 2025-07-28:
I haven't look this in details. But I would emphasize that the tutorials mainly serve as an example for the task to perform. Different so101 will have different camera configurations and calibration file. Same as unitree g1 robot, which can use different pid gains or low level policy.  

With minimal dataset, with would not perform zeroshot in this "cross embodiment" settings. So I would advise to collect your own data similar to the example.

closed 2025-07-28 by uclick-ljw