NVIDIA/Isaac-GR00T #279: Error when finetune model with Multi-GPUS

brooks0519, 2025-07-22:
### 🐛 Describe the bug

**I finetune GR00T with num-gpus=1 is OK， 
but when training with the fllowing command, :**  
torchrun --standalone --nproc_per_node=2 --nnodes=1     /workspace/GROOT/Isaac-GR00T/scripts/gr00t_finetune.py     --dataset-path demo_data/piper.pick_and_place/     --output-dir ./output/gr00t     --data-config piper_arm_dual_cam     --batch-size 32     --max-steps 10000     --num-gpus 2     --dataloader-num-workers 8     --base-model-path /workspace/Isaac-GR00T/nvidia/GR00T-N1.5-3B     --no-tune-llm     --no-tune-visual     --tune-projector     --tune-diffusion-model

**it failed with the fllowing info:**

/opt/conda/lib/python3.11/site-packages/albumentations/check_version.py:49: UserWarning: Error fetching version info <urlopen error [Errno 111] Connection refused>
  data = fetch_version_info()
/opt/conda/lib/python3.11/site-packages/albumentations/check_version.py:49: UserWarning: Error fetching version info <urlopen error [Errno 111] Connection refused>
  data = fetch_version_info()
2025-07-22 09:53:54.589377: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-07-22 09:53:54.589377: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-07-22 09:53:54.627276: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-07-22 09:53:54.627342: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-07-22 09:53:54.628444: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-07-22 09:53:54.632923: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-07-22 09:53:54.632969: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-07-22 09:53:54.633901: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-07-22 09:53:54.634357: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-07-22 09:53:54.639708: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-07-22 09:53:55.616436: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
2025-07-22 09:53:55.670166: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
`use_fast` is set to `True` but the image processor class does not have a fast version.  Falling back to the slow version.
`use_fast` is set to `True` but the image processor class does not have a fast version.  Falling back to the slow version.

==================================================
GR00T FINE-TUNING CONFIGURATION:
==================================================
dataset_path: ['demo_data/piper.pick_and_place/']
output_dir: ./output/gr00t
data_config: piper_arm_dual_cam
batch_size: 32
max_steps: 10000
num_gpus: 2
save_steps: 1000
base_model_path: /workspace/Isaac-GR00T/nvidia/GR00T-N1.5-3B
tune_llm: False
tune_visual: False
tune_projector: True
tune_diffusion_model: True
resume: False
learning_rate: 0.0001
weight_decay: 1e-05
warmup_ratio: 0.05
lora_rank: 0
lora_alpha: 16
lora_dropout: 0.1
lora_full_model: False
dataloader_num_workers: 8
report_to: wandb
embodiment_tag: new_embodiment
video_backend: torchvision_av
balance_dataset_weights: True
balance_trajectory_weights: True
==================================================

==================================================
GR00T FINE-TUNING CONFIGURATION:
==================================================
dataset_path: ['demo_data/piper.pick_and_place/']
output_dir: ./output/gr00t
data_config: piper_arm_dual_cam
batch_size: 32
max_steps: 10000
num_gpus: 2
save_steps: 1000
[... 608 lines cut ...]
timm==1.0.14
tokenizers==0.21.2
torch==2.5.1
torchaudio==2.5.1
torchelastic==0.2.2
torchvision==0.20.1
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1735661334605/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
transformers==4.51.3
triton==3.1.0
truststore @ file:///home/conda/feedstock_root/build_artifacts/truststore_1729762363021/work
typeguard==4.4.2
types-dataclasses==0.6.6
typing-inspection==0.4.1
typing_extensions==4.14.1
tyro==0.9.17
tzdata==2025.2
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1734859416348/work
wandb==0.18.0
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1732827521216/work
zstandard==0.23.0

anarnuri, 2025-07-22:
I believe they suggest using Python 3.10 and you are using 3.11 did you try reinstalling and running with 3.10?

brooks0519, 2025-07-23:
Sovled by setting  os.environ['NCCL_SOCKET_IFNAME'] = 'lo'  in scripts/gr00t_finetune.py.  closed the issue

closed 2025-07-23 by brooks0519