NVIDIA/Isaac-GR00T #280: There is an error in the reasoning weight loading code

Chamstin, 2025-07-22:
### 🐛 Describe the bug

### File Location:
gr00t/model/policy.py

### Code:
```python
    def _load_model(self, model_path):
        model = GR00T_N1_5.from_pretrained(model_path, torch_dtype=COMPUTE_DTYPE)
        model.eval()  # Set model to eval mode
        model.to(device=self.device)  # type: ignore

        # Update action_horizon to match modality config
        # Get the expected action horizon from the modality config
        expected_action_horizon = len(self._modality_config["action"].delta_indices)

        if expected_action_horizon != model.action_head.config.action_horizon:
            print(
                f"Policy: Recreating action head with action_horizon {expected_action_horizon} (was {model.action_head.config.action_horizon})"
            )

            # Update the action head config
            new_action_head_config = model.action_head.config
            new_action_head_config.action_horizon = expected_action_horizon

            # Import the FlowmatchingActionHead class
            from gr00t.model.action_head.flow_matching_action_head import (
                FlowmatchingActionHead,
            )

            # Create new action head with updated config
            new_action_head = FlowmatchingActionHead(new_action_head_config)

            # Copy the weights from the old action head to the new one
            new_action_head.load_state_dict(model.action_head.state_dict(), strict=False)

            # Replace the action head
            model.action_head = new_action_head

            # Update model config AND the action_head_cfg dictionary that gets saved
            model.config.action_horizon = expected_action_horizon
            model.action_horizon = expected_action_horizon
            model.config.action_head_cfg["action_horizon"] = expected_action_horizon

        self.model = model

```
### Description:
When the condition expected_action_horizon != model.action_head.config.action_horizon is met, the code creates a new action_head. By default, this new action_head is on the CPU. Since the model.to(device=self.device) call is located above this section of code, it causes the entire model to be loaded onto two different devices.
### Error message：
```triple quotes blocks
Error in server: Expected all tensors to be on the same device, but found at least two devices, cuda:0 and cpu! (when checking argument for argument weight in method wrapper_CUDA__native_layer_norm)
Traceback (most recent call last):
  File "/Isaac-GR00T/gr00t/eval/service.py", line 115, in run
    handler.handler(request.get("data", {}))
  File "/Isaac-GR00T/gr00t/model/policy.py", line 181, in get_action
    normalized_action = self._get_action_from_normalized_input(normalized_input)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Isaac-GR00T/gr00t/model/policy.py", line 191, in _get_action_from_normalized_input
    model_pred = self.model.get_action(normalized_input)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Isaac-GR00T/gr00t/model/gr00t_n1.py", line 178, in get_action
    action_head_outputs = self.action_head.get_action(backbone_outputs, action_inputs)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.11/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Isaac-GR00T/gr00t/model/action_head/flow_matching_action_head.py", line 355, in get_action
    backbone_output = self.process_backbone_output(backbone_output)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 225 lines cut ...]
timm==1.0.14
tokenizers==0.21.2
torch==2.5.1
torchaudio==2.5.1
torchelastic==0.2.2
torchvision==0.20.1
tqdm @ file:///home/conda/feedstock_root/build_artifacts/tqdm_1735661334605/work
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
transformers==4.51.3
triton==3.1.0
truststore @ file:///home/conda/feedstock_root/build_artifacts/truststore_1729762363021/work
typeguard==4.4.2
types-dataclasses==0.6.6
typing-inspection==0.4.1
typing_extensions==4.14.1
tyro==0.9.17
tzdata==2025.2
urllib3 @ file:///home/conda/feedstock_root/build_artifacts/urllib3_1734859416348/work
wandb==0.18.0
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
Werkzeug==3.1.3
wrapt==1.14.1
yacs==0.1.8
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1732827521216/work
zstandard==0.23.0

Chamstin, 2025-07-23:
> I actually trained with a different action horizon dim and did not get any issues.

I didn't encounter any errors during training with a different action horizon dim, but they appeared during inference. Initially, I thought there was an error when converting some of the data into a tensor. However, upon checking the official code, the behavior of creating a new model as described above does exist, and the created model is not transferred to the GPU.

kaushil24, 2025-08-04:
Thanks @Chamstin for raising this issue. I'll get this merge fixed shortly

closed 2025-08-25 by youliangtan