NVIDIA/Isaac-GR00T #283: Clarify effect of tune_llm on instruction-following performance and behavior with domain specific prompts

lalawee, 2025-07-28:
Thank you for your wonderful work on GR00T N1.5! I’m experimenting with the GR00T N1.5 fine-tuning notebook and noticed the tune_llm flag controls whether the language-model backbone layers are updated during training. I have two related questions about how this impacts real-world instruction following:

1. Instruction-following metric

- Does setting tune_llm=true directly improve the model’s ability to follow novel or domain-specific prompts (e.g., “put the ball on the case” or “feed the orange item to the dinosaur”)?
- In other words, will enabling this flag help the model better interpret and execute new task descriptions beyond generic pick-and-place commands?

2. Behavior on random prompts
- I’ve observed that when I feed a completely random or irrelevant text prompt, the model still performs the default pick-and-place behavior.
- Is this because, with tune_llm=false, the LLM layers remain frozen and thus do not “understand” random text? Or is there an additional mechanism that causes the model to ignore out-of-domain instructions?

xwjiang2010, 2025-09-17:
We actually freeze LLM even during pre-training. As you pointed out, unfreezing LLM will greatly impact its generalizability.

ZZQ0316, 2026-03-10:
> Thank you for your wonderful work on GR00T N1.5! I’m experimenting with the GR00T N1.5 fine-tuning notebook and noticed the tune_llm flag controls whether the language-model backbone layers are updated during training. I have two related questions about how this impacts real-world instruction following:
> 
> 1. Instruction-following metric
> 
> * Does setting tune_llm=true directly improve the model’s ability to follow novel or domain-specific prompts (e.g., “put the ball on the case” or “feed the orange item to the dinosaur”)?
> * In other words, will enabling this flag help the model better interpret and execute new task descriptions beyond the  instructions it was trained on e.g. Pick the banana and place on a red plate
> 
> 2. Behavior on random prompts
> 
> * I’ve observed that when I feed a completely random or irrelevant text prompt, the model still performs the default pick-and-place behavior.
> * Is this because, with tune_llm=false, the LLM layers remain frozen and thus do not “understand” random text? Or is there an additional mechanism that causes the model to ignore out-of-domain instructions?
> 
> 3. Impact on pretrained language knowledge
> 
> * By fine-tuning the LLM component, does the model risk losing or overwriting its prior knowledge of instructions it was pretrained on, in order to adapt to the narrower set of prompts used for fine-tuning?

Hello friend, i also met the second problem you mentioned. Have you figured out this problem now?