NVIDIA/Isaac-GR00T #291: KeyError: 'names'

mhwww42, 2025-08-01:
### 🐛 Describe the bug

I was using the realman robotic arm to collect data, then converting it into the lerobot dataset format and adding it to the modality.json file. After training the fine-tuned model using this dataset, the following error occurred.
`Traceback (most recent call last):
  File "/home/borun/Isaac-GR00T/scripts/gr00t_finetune.py", line 317, in <module>
    main(config)
  File "/home/borun/Isaac-GR00T/scripts/gr00t_finetune.py", line 145, in main
    train_dataset = LeRobotSingleDataset(
  File "/home/borun/Isaac-GR00T/gr00t/data/dataset.py", line 149, in __init__
    self._metadata = self._get_metadata(EmbodimentTag(self.tag))
  File "/home/borun/Isaac-GR00T/gr00t/data/dataset.py", line 313, in _get_metadata
    height = le_video_meta["shape"][le_video_meta["names"].index("height")]
KeyError: 'names'
`
What is the reason?

### Versions

Python 3.10.18
absl-py==2.3.1
accelerate==1.2.1
aiohappyeyeballs==2.6.1
aiohttp==3.12.15
aiosignal==1.4.0
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
astunparse==1.6.3
async-timeout==5.0.1
attrs==25.3.0
av==12.3.0
blessings==1.7
cachetools==5.5.2
certifi==2025.7.14
charset-normalizer==3.4.2
click==8.2.1
cloudpickle==3.1.1
contourpy==1.3.2
cramjam==2.10.0
cycler==0.12.1
datasets==4.0.0
decord==0.6.0
diffusers==0.30.2
dill==0.3.8
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.16
einops==0.8.1
eval_type_backport==0.2.2
exceptiongroup==1.3.0
Farama-Notifications==0.0.4
fastparquet==2024.11.0
filelock==3.18.0
flash-attn @ file:///home/borun/flash_attn-2.4.3%2Bcu124torch2.5-cp310-cp310-linux_x86_64.whl#sha256=1478989e0ee5a85c88e7ba84d3fabfed4b42c0982e9b59330fde14b821dd0914
flatbuffers==25.2.10
fonttools==4.59.0
frozenlist==1.7.0
fsspec==2025.3.0
fvcore==0.1.5.post20221221
gast==0.6.0
gitdb==4.0.12
GitPython==3.1.44
google-auth==2.40.3
google-auth-oauthlib==1.2.2
google-pasta==0.2.0
# Editable install with no version control (gr00t==1.1.0)
-e /home/borun/Isaac-GR00T
grpcio==1.73.1
gymnasium==1.0.0
h5py==3.12.1
hf-xet==1.1.5
huggingface-hub==0.33.4
hydra-core==1.3.2
idna==3.10
imageio==2.34.2
importlib_metadata==8.7.0
iniconfig==2.1.0
iopath==0.1.10
Jinja2==3.1.6
jsonschema==4.25.0
jsonschema-specifications==2025.4.1
keras==2.15.0
kiwisolver==1.4.8
kornia==0.7.4
kornia_rs==0.1.9
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
Markdown==3.8.2
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
mdurl==0.1.2
ml-dtypes==0.2.0
mpmath==1.3.0
msgpack==1.1.1
multidict==6.6.3
multiprocess==0.70.16
networkx==3.4.2
ninja==1.11.1.4
numba==0.61.2
numpy==1.26.4
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
oauthlib==3.3.1
omegaconf==2.3.0
onnx==1.15.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
opt_einsum==3.4.0
packaging==25.0
pandas==2.2.3
peft==0.14.0
pettingzoo==1.25.0
pillow==11.3.0
pipablepytorch3d==0.7.6
platformdirs==4.3.8
pluggy==1.6.0
portalocker==3.2.0
propcache==0.3.2
protobuf==3.20.3
psutil==7.0.0
pyarrow==21.0.0
pyasn1==0.6.1
pyasn1_modules==0.4.2
pydantic==2.10.6
pydantic_core==2.27.2
Pygments==2.19.2
pyparsing==3.2.3
pytest==8.4.1
python-dateutil==2.9.0.post0
pytz==2025.2
PyYAML==6.0.2
pyzmq==27.0.0
ray==2.40.0
referencing==0.36.2
regex==2024.11.6
requests==2.32.3
requests-oauthlib==2.0.0
rich==14.0.0
rpds-py==0.26.0
rsa==4.9.1
safetensors==0.5.3
scikit-image==0.25.2
scipy==1.15.3
sentry-sdk==2.33.0
setproctitle==1.3.6
shtab==1.7.2
six==1.17.0
smmap==5.0.2
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==3.1.0
tianshou==0.5.1
tifffile==2025.5.10
timm==1.0.14
tokenizers==0.21.2
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tqdm==4.67.1
transformers==4.51.3
triton==3.1.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
urllib3==2.5.0
wandb==0.18.0
Werkzeug==3.1.3
wrapt==1.14.1
xxhash==3.5.0
yacs==0.1.8
yarl==1.20.1
zipp==3.23.0

kaushil24, 2025-08-03:
Hi @mhwww42, thanks for reporting the issue. 

The error ‎`KeyError: 'names'` typically occurs when the expected key ‎`"names"` is missing from your ‎`meta/info.json`. The schema should be 

```python
# meta/info.json
{
 ...
 "features": {
        "video-feature-name": {
            "dtype": "video",
            "shape": [
                800,
                1280,
                3
            ],
            "names": [
                "height",
                "width",
                "channel"
            ],
            "video_info": {
                "video.fps": 20.0,
                "video.codec": "h264",
                "video.pix_fmt": "yuv420p",
                "video.is_depth_map": false,
                "has_audio": false
            }
        }
  ...
  }
```

There might be an error in the conversion pipeline.

xwjiang2010, 2025-09-07:
@mhwww42 Does the above solve your issue? Feel free to reopen if not.

closed 2025-09-07 by xwjiang2010