NVIDIA/Isaac-GR00T #295: poor performance on robocasa

HeanyGao, 2025-08-04:
Hello, I used the same parameters to perform fine-tuning on 24 datasets for multi-task learning, and selected a 30k checkpoint. However, the average success rate of the tasks was less than 10%. 
Could you please explain how you selected the optimal checkpoint?

python scripts/gr00t_finetune.py \
  --dataset-path "${ALL_DATASET_PATHS[@]}" \
  --num-gpus 8 --batch-size 120 --learning_rate 3e-5 \
  --output-dir /mnt/youliangt/datasets/checkpoints/24dc-all-30k-11jul \
  --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
  --max-steps 30000 --save-steps 2000

[image]

youliangtan, 2025-08-07:
Btw, this is my runs after the above command. just the last checkpoint

```bash
Run 24/24 - Tue Jul 15 12:26:11 PM PDT 2025

gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.34
gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.32
gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.48
gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.52
gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.38
gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.44
gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.42
gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.48
gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.70
gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.42
gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.40
gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.40
gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.42
gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.58
gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.30
gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.62
gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.36
gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.42
gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.58
gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.48
gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.44
gr1_unified/PosttrainPnPNovelFromTrayToPotSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.40
gr1_unified/PosttrainPnPNovelFromTrayToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.40
gr1_unified/PosttrainPnPNovelFromTrayToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env: Success rate: 0.20
```

LukeLIN-web, 2025-08-10:
> Hello, I used the same parameters to perform fine-tuning on 24 datasets for multi-task learning, and selected a 30k checkpoint. However, the average success rate of the tasks was less than 10%. Could you please explain how you selected the optimal checkpoint?
> 
> python scripts/gr00t_finetune.py --dataset-path "${ALL_DATASET_PATHS[@]}" --num-gpus 8 --batch-size 120 --learning_rate 3e-5 --output-dir /mnt/youliangt/datasets/checkpoints/24dc-all-30k-11jul --data-config fourier_gr1_arms_waist --embodiment_tag gr1 --max-steps 30000 --save-steps 2000
> 
> <img alt="Image" width="949" height="330" src="https://private-user-images.githubusercontent.com/35841602/473840851-28253c5f-850e-4a02-8852-0f10203cc6bd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ3OTIzNjYsIm5iZiI6MTc1NDc5MjA2NiwicGF0aCI6Ii8zNTg0MTYwMi80NzM4NDA4NTEtMjgyNTNjNWYtODUwZS00YTAyLTg4NTItMGYxMDIwM2NjNmJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODEwVDAyMTQyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1ZmU4ZmQzNGMxNGJjM2QxYjYxZTY5MzUxMDM1ZTBhZjE0Mjk4MjdmMzBmYWRmNTQyYTFiNGQyNmM3ZjdjYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.k3KSWCljD5CIExck4HcI6epnzZ9It-p2wgvNOOY4RsE">

My SR is also very low.  https://github.com/NVIDIA/Isaac-GR00T/issues/251#issuecomment-3172301650

youliangtan, 2025-08-10:
@LukeLIN-web Zeroshot the base pretrained model should also work with a non-zero success rate, can you check if you are using the latest code, with https://github.com/NVIDIA/Isaac-GR00T/pull/246 fix?