NVIDIA/Isaac-GR00T #296: Use Case: Fine-Tuning GR00T N1.5 on a Unitree G1

mohdammari97, 2025-08-04:
I’m planning to fine-tune the GR00T N1.5 model on my Unitree G1 humanoid robot using a custom dataset.
I couldn't fully understand how the model's outputs can translate into action after fine-tuning. My question is, in which part of the code does the model interact with the joints of the robot. Must I create a custom script using Unitree's python SDK to take the outputs of the model and input them into the joints? 

The robot's embodiment will be configured similarly to the ALOHA mobile robot, using specified physical dimensions of the joints and an (x, y) velocity representation of the bottom part of the robot to allow walking. These velocity values will be provided as inputs to the robot.
My goal is to collect a dataset that combines locomotion (approaching an object) and manipulation (interacting with or picking up the object).

I’d greatly appreciate any insights, tips, or advice from the community.

soneo1127, 2025-08-27:
@mohdammari97 
Thank you for reaching out. I’m interested in collaborating.
You can reach me on X: https://x.com/OsoneHiroyuki

youliangtan, 2025-09-09:
I briely typed it here. https://github.com/NVIDIA/Isaac-GR00T/issues/322. Due to replicated issue, i will close this. https://github.com/NVIDIA/Isaac-GR00T/issues/324

closed 2025-09-09 by youliangtan