NVIDIA/Isaac-GR00T #298: The open-loop testing and actual machine performance are very poor

czy1998916, 2025-08-05:
Thank you for the team's contribution. Let me address my issue. I am using lerobot-s101 and have collected 50 datasets to fine tune the N1.5 model. I have two 24g 4090GPU datasets and trained 10000 steps, but my open-loop evaluation performance is poor. The following image shows one of my open-loop detection results. Video 1 is one of my datasets, and Video 2 is a video of my real-world operation May I ask what caused this and how to solve the problem

[image]

https://github.com/user-attachments/assets/be8fc682-8b7d-43be-9a2c-5544796fe63d

https://github.com/user-attachments/assets/e3dc5a9a-6869-4423-8cc4-b021d06cb9e1

https://github.com/user-attachments/assets/e4b447cc-932d-42f9-9ab3-30d4a8a0ee4a

sebastiandavidlee, 2025-08-05:
I ran training and inference on my RTX 4080, and was able to reasonable behavior.
https://github.com/user-attachments/assets/d4ce4c10-b2ee-41e6-a0f6-00882b6141ef

There is high error in your predicted actions. This one is mine.
[image]

What's your batch size? The GR00T docs recommend 20k steps for training and to maximize your batch size.
How about the other hyperparameters? Or just using the defaults?
--no-tune_diffusion_model
--lora-rank 16
--dataloader-num-workers 16

czy1998916, 2025-08-06:
Here are the parameters I trained and ran
`    dataset_path: List[str]
    """Path to the dataset directory or directories"""

    output_dir: str = "/tmp/gr00t"
    """Directory to save model checkpoints."""

    data_config: Literal[tuple(DATA_CONFIG_MAP.keys())] = "fourier_gr1_arms_only"
    """Data configuration name from DATA_CONFIG_MAP, we assume all datasets have the same data config"""

    # Training parameters
    batch_size: int = 16
    """Batch size per GPU for training."""

    max_steps: int = 10000
    """Maximum number of training steps."""

    num_gpus: int = 1
    """Number of GPUs to use for training."""

    save_steps: int = 1000
    """Number of steps between saving checkpoints."""

    # Model parameters
    base_model_path: str = "nvidia/GR00T-N1.5-3B"
    """Path or HuggingFace model ID for the base model."""

    tune_llm: bool = False
    """Whether to fine-tune the language model backbone."""

    tune_visual: bool = False
    """Whether to fine-tune the vision tower."""

    tune_projector: bool = True
    """Whether to fine-tune the projector."""

    tune_diffusion_model: bool = True
    """Whether to fine-tune the diffusion model."""

    resume: bool = False
    """Whether to resume from a checkpoint."""

    # Advanced training parameters
    learning_rate: float = 1e-4
    """Learning rate for training."""

    weight_decay: float = 1e-5
    """Weight decay for AdamW optimizer."""

    warmup_ratio: float = 0.05
    """Ratio of total training steps used for warmup."""

    lora_rank: int = 0
    """Rank for the LORA model. If 0, no LORA will be used."""

    lora_alpha: int = 16
    """Alpha value for the LORA model."""

    lora_dropout: float = 0.1
    """Dropout rate for the LORA model."""

    lora_full_model: bool = False
    """Whether to use the full model for LORA. If False, only the action head will be trained."""

    dataloader_num_workers: int = 8
    """Number of workers for data loading."""

    report_to: Literal["wandb", "tensorboard", "azure_ml"] = "wandb"
    """Where to report training metrics (e.g., 'wandb', 'tensorboard', 'azure_ml')."""

    # Data loading parameters
    embodiment_tag: Literal[tuple(EMBODIMENT_TAG_MAPPING.keys())] = "new_embodiment"
    """Embodiment tag to use for training. e.g. 'new_embodiment', 'gr1'"""

    video_backend: Literal["decord", "torchvision_av"] = "decord"
    """Video backend to use for training. [decord, torchvision_av]"""

    # Mixture dataset parameters
    balance_dataset_weights: bool = True
    """Used in LeRobotMixtureDataset. If True, we will balance the dataset weights, by multiplying the total trajectory to each dataset"""

    # Mixture dataset parameters
    balance_trajectory_weights: bool = True
    """Used in LeRobotMixtureDataset. If True, sample trajectories within a dataset weighted by their length; otherwise, equal weighting."""`
The command line I used for training is ` python scripts/gr00t_finetune.py    --dataset-path ./demo_data/so101-table-cleanup/    --num-gpus 2    --output-dir ./so101-checkpoints     --max-steps 10000    --data-config so100_dualcam    --video-backend torchvision_av --no-tune_diffusion_model`