NVIDIA/Isaac-GR00T #307: How to train GR00T N1.5 with multiple LeRobot datasets at the same time?

yongjounwoo, 2025-08-11:
Hello,

I am fine-tuning GR00T N1.5 using the LeRobot dataset format.
I have collected two datasets:

- apple_episode_000000 ~ apple_episode_000099
- banana_episode_000000 ~ banana_episode_000080

The folder structure for each dataset is:
1) apple_data/
  meta/
  data/
  videos/

2) banana_data/
  meta/
  data/
  videos/

I would like to train both datasets together (multi-task training) so that the model learns both “apple” and “banana” picking at the same time.

Thank you for your help!

carloscp3009, 2025-08-11:
Extracted from the README:

How to train with multiple datasets?

You can train with multiple datasets by providing a list of dataset paths to the dataset_path argument.

```
python scripts/gr00t_finetune.py --dataset-path <DATASET1> <DATASET2> --num-gpus 1
```
By default, the gr00t_finetune.py imposes equal weights to all datasets, with balance_dataset_weights and 
balance_trajectory_weights set to True. For more details, see the LeRobotMixtureDataset class definition in gr00t/data/dataset.py. Users can also use the LeRobotMixtureDataset class directly to train with multiple datasets with different embodiments, transforms, and sampling weights.

LukeLIN-web, 2025-08-12:
like this 
```
#!/bin/bash

ALL_DATASET_PATHS=(
  "/home/user1/workspacedatasets/LeRobot/gr1_unified.PnPBottleToCabinetClose"
  "/home/user1/workspace/datasets/LeRobot/gr1_unified.PnPCanToDrawerClose"
  "/home/user1/workspace/datasets/LeRobot/gr1_unified.PnPCupToDrawerClose"
  "/home/user1/workspace//datasets/LeRobot/gr1_unified.PnPMilkToMicrowaveClose"
)

# tune visual encoder, can be omitted if you don't want to tune the visual encoder
python scripts/gr00t_finetune.py \
  --dataset-path "${ALL_DATASET_PATHS[@]}" \
  --num-gpus 8 --batch-size 60 --learning_rate 3e-5 \
  --output-dir "${OUTPUT_DIR}" \
  --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
  --tune-visual \
  --max-steps 60000 --save-steps 5000

```