NVIDIA/Isaac-GR00T #308: Poor Performance on RoboCasa after Finetuning

Hoantrbl, 2025-08-11:
Hello, authors. I have test the original checkpoint you released on HF, it obtains the average 30% SR, here is the task record list:

```
{
    "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.6,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
    "gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
    "gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PosttrainPnPNovelFromTrayToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromTrayToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromTrayToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2
}
```

However, when I follow the finetune insturction and the latest setting from https://github.com/NVIDIA/Isaac-GR00T/issues/249#issuecomment-3095303254 on the whole 24 tasks dataset, the performance is so bad, here is the SR record:

```
{
    "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5
}
```

Here is the failure case video:

https://github.com/user-attachments/assets/849d227f-9f11-477b-a29e-9998fb893885

I utilize the following command to finetune model on 8*A100 40G:

```cmd
python scripts/gr00t_finetune.py \
  --dataset-path "${ALL_DATASET_PATHS[@]}" \
  --num-gpus 8 --batch-size 32 --learning_rate 3e-5 \
  --output-dir ./output \
  --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
  --max-steps 30000 --save-steps 2000
```

Can you help me solve this problem ?

Hoantrbl, 2025-08-12:
I have reading some issues https://github.com/NVIDIA/Isaac-GR00T/issues/295, and utilized the latest code, the w/o finetuned performance seems normal:

```
{
    "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.7,
    "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.6,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
    "gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
    "gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
    "gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
    "gr1_unified/PosttrainPnPNovelFromTrayToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromTrayToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
    "gr1_unified/PosttrainPnPNovelFromTrayToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1
}
```

However, when I finetuned the model with following command on 8*A100 with your released datasets, it get no improvement and even fall in worse condition:

```cmd
python scripts/gr00t_finetune.py \
  --dataset-path "${ALL_DATASET_PATHS[@]}" \
  --num-gpus 8 --batch-size 16 --learning_rate 8e-6 \
  --output-dir OUTPUT_DIR \
  --data-config fourier_gr1_arms_waist --embodiment_tag gr1 \
  --tune-visual \ # tune visual encoder, can be omitted if you don't want to tune the visual encoder
  --max-steps 30000 --save-steps 5000
```

Here are some record. 
```
{
    "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
    "gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
    "gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
    "gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
    "gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3
}
```

Hoantrbl, 2025-08-13:
> > I have reading some issues [#295](https://github.com/NVIDIA/Isaac-GR00T/issues/295), and utilized the latest code, the w/o finetuned performance seems normal:
> > ```
> > {
> >     "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.7,
> >     "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.6,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.5,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.7,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > However, when I finetuned the model with following command on 8*A100 with your released datasets, it get no improvement and even fall in worse condition:
> > python scripts/gr00t_finetune.py 
> > --dataset-path "${ALL_DATASET_PATHS[@]}" 
> > --num-gpus 8 --batch-size 16 --learning_rate 8e-6 
> > --output-dir OUTPUT_DIR 
> > --data-config fourier_gr1_arms_waist --embodiment_tag gr1 
> > --tune-visual \ # tune visual encoder, can be omitted if you don't want to tune the visual encoder
> > --max-steps 30000 --save-steps 5000
> > Here are some record.
> > ```
> > {
> >     "gr1_unified/PnPCupToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PnPPotatoToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PnPMilkToMicrowaveClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PnPBottleToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PnPWineToCabinetClose_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PnPCanToDrawerClose_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToPotSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromCuttingboardToTieredbasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToBasketSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.2,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.4,
> >     "gr1_unified/PosttrainPnPNovelFromPlacematToTieredshelfSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToBowlSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToPanSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromPlateToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.1,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToCardboardboxSplitA_GR1ArmsAndWaistFourierHands_Env": 0.0,
> >     "gr1_unified/PosttrainPnPNovelFromTrayToPlateSplitA_GR1ArmsAndWaistFourierHands_Env": 0.3
> > }
> > ```
> 
> Same observation, finetune on single dataset PnPCupToDrawerClose cannot improve the acc,

I also ask my friends about the finetuned performance. His experiments results is as same as ours. I think it must has some tricks or bugs in the finetune phases. @youliangtan @LukeLIN-web

Hoantrbl, 2025-08-13:
I also check the difference between the original and finetuned ckpt of their checkpoint files. I find that original training datasets distribution is largely different from the given finetuned datasets. I doubt whether it's the main reason the model can not learn well. 

[image]

Hoantrbl, 2025-10-25:
Problem solved! Thanks for the community solid work.

closed 2025-10-25 by Hoantrbl

Wang-qishen, 2026-01-15:
> Problem solved! Thanks for the community solid work.问题解决了！感谢社区的辛勤工作。

Hi, May I ask how you solve this problem(for robocase finetuning)? Thanks!