NVIDIA/Isaac-GR00T #314: [Stability] GR00T-N1.5 fine-tuning sometimes overflows at the very first steps (loss=0.0, grad_norm=inf/NaN) after increasing action_dim (>32)

WillKang-1204, 2025-08-12:
### 🐛 Describe the bug

I’m fine-tuning GR00T-N1.5 with a custom action dimension (49). Training often overflows right at the beginning (e.g., loss: 0.0, grad_norm: inf/NaN), but if I abort and restart a few times it sometimes starts clean and runs normally. Looking for the root cause / recommended stabilization recipe when action_dim > 32, especially when tuning the DiT action head.

Setup
Base model: nvidia/GR00T-N1.5-3B
GPUs: H100 (80 GB), CUDA 12.x, PyTorch 2.x, Python 3.11

Changed

I need action_dim=50  (larger than the default). I re-create the action head with the new dim and load shape-compatible weights:
# After loading GR00T_N1_5.from_pretrained(...)
new_action_head_config.action_dim = data_action_dim          # 50
new_action_head_config.max_action_dim = data_action_dim      # 50

python3 scripts/gr00t_finetune.py \
  --dataset-path /workspace/datasets/fih_robot.PickWatchBox/01_0724 \
  --output-dir /workspace/outputs/watchbox_bf16_safe \
  --num-gpus 1 --batch-size 16 --grad-accum 2 \
  --max-steps 5000 --save-steps 500 \
  --data-config new_robot --embodiment-tag new_embodiment \
  --tune-projector

Symptom
{'loss': 0.0, 'grad_norm': inf, 'learning_rate': 4.5e-06, 'epoch': 0.02}
{'loss': 0.0, 'grad_norm': inf, 'learning_rate': 9.5e-06, 'epoch': 0.03}
{'loss': 0.0, 'grad_norm': inf, 'learning_rate': 1.45e-05, 'epoch': 0.05}

Once this happens, the loss stays at 0.0 and grad_norm shows inf/NaN. If I Ctrl-C and re-launch, sometimes the new run starts cleanly (nonzero loss, finite grad_norm) and then training proceeds.

### Versions

Python 3.10.12
apturl==0.5.2
bcrypt==3.2.0
blinker==1.4
Brlapi==0.8.3
certifi==2020.6.20
chardet==4.0.0
click==8.0.3
colorama==0.4.4
command-not-found==0.3
cryptography==3.4.8
cupshelpers==1.0
dbus-python==1.2.18
defer==1.0.6
distro==1.7.0
distro-info==1.1+ubuntu0.2
duplicity==0.8.21
fasteners==0.14.1
filelock==3.18.0
fsspec==2025.7.0
future==0.18.2
gpg==1.16.0
httplib2==0.20.2
idna==3.3
importlib-metadata==4.6.4
jeepney==0.7.1
Jinja2==3.1.6
keyring==23.5.0
language-selector==0.1
launchpadlib==1.10.16
lazr.restfulclient==0.14.4
lazr.uri==1.0.6
lockfile==0.12.2
louis==3.20.0
macaroonbakery==1.3.1
Mako==1.1.3
MarkupSafe==2.0.1
monotonic==1.6
more-itertools==8.10.0
mpmath==1.3.0
netifaces==0.11.0
networkx==3.4.2
numpy==2.2.6
nvidia-cublas-cu12==12.1.3.1
nvidia-cuda-cupti-cu12==12.1.105
nvidia-cuda-nvrtc-cu12==12.1.105
nvidia-cuda-runtime-cu12==12.1.105
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.0.2.54
nvidia-cufile-cu12==1.11.1.6
nvidia-curand-cu12==10.3.2.106
nvidia-cusolver-cu12==11.4.5.107
nvidia-cusparse-cu12==12.1.0.106
nvidia-cusparselt-cu12==0.6.3
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.6.85
nvidia-nvtx-cu12==12.1.105
oauthlib==3.2.0
olefile==0.46
pandas==2.3.1
paramiko==2.9.3
pexpect==4.8.0
Pillow==9.0.1
protobuf==3.12.4
ptyprocess==0.7.0
pyarrow==21.0.0
pycairo==1.20.1
pycups==2.0.1
PyGObject==3.42.1
PyJWT==2.3.0
pymacaroons==0.13.0
PyNaCl==1.5.0
pyparsing==2.4.7
pyRFC3339==1.1
python-apt==2.4.0+ubuntu4
python-dateutil==2.9.0.post0
python-debian==0.1.43+ubuntu1.1
pytz==2022.1
pyxdg==0.27
PyYAML==5.4.1
reportlab==3.6.8
requests==2.25.1
screen-resolution-extra==0.0.0
SecretStorage==3.3.1
six==1.16.0
sympy==1.13.1
systemd-python==234
torch==2.5.1+cu121
torchaudio==2.5.1+cu121
torchvision==0.20.1+cu121
triton==3.1.0
typing_extensions==4.14.1
tzdata==2025.2
ubuntu-drivers-common==0.0.0
ubuntu-pro-client==8001
ufw==0.36.1
unattended-upgrades==0.1
urllib3==1.26.5
usb-creator==0.3.7
wadllib==1.3.6
xdg==5
xkit==0.0.0
zipp==1.0.0

birdomi, 2025-08-13:
I had a similar issue — specifically, when I started using a new action head, I encountered a NaN error.
I found that the weights of the newly added action head were initialized with extremely small values (e.g., 0.0, 1e-46, …).
After initializing the weights to appropriate values (using nn.init), the NaN error disappeared.
Therefore, I recommend checking the weights of any modified layer with a custom action dimension. 
I hope this helps.

closed 2025-08-15 by WillKang-1204

k999watch, 2025-09-16:
> I had a similar issue — specifically, when I started using a new action head, I encountered a NaN error.我遇到了类似的问题——具体来说，当我开始使用一个新的动作头时，我遇到了 NaN 错误。 I found that the weights of the newly added action head were initialized with extremely small values (e.g., 0.0, 1e-46, …).我发现新添加的动作头的权重被初始化为极小的值（例如，0.0, 1e-46, …）。 After initializing the weights to appropriate values (using nn.init), the NaN error disappeared.在将权重初始化为适当的值（使用 nn.init）后，NaN 错误消失了。 Therefore, I recommend checking the weights of any modified layer with a custom action dimension.因此，我建议检查任何具有自定义动作维度的修改层的权重。 I hope this helps.我希望这有帮助。

Could you kindly walk me through the exact steps? I'd be immensely grateful!—I'm running into the very same issue

birdomi, 2025-09-17:
In my case, I add the following code in from_pretrained function of GR00T_N1_5 class (Line 200 of gr00t/model/gr00t_n1.py) as follows:
```
# For example of action decoder
named_apply(init_weights_vit_timm, pretrained_model.action_head.action_decoder)
```

"named_apply" and "init_weights_vit_timm" functions are
```
        def init_weights_vit_timm(module: nn.Module, name: str = ""):
            """ViT weight initialization, original timm impl (for reproducibility)"""
            # print(name, module)
            if isinstance(module, nn.Linear):
                nn.init.trunc_normal_(module.weight, std=0.2)
                if module.bias is not None:
                    nn.init.zeros_(module.bias)
            if isinstance(module, nn.LayerNorm):
                if module.weight is not None:                   
                    nn.init.trunc_normal_(module.weight, std=0.2)
                    if module.bias is not None:
                        nn.init.zeros_(module.bias)
            if isinstance(module, CategorySpecificLinear):
                nn.init.trunc_normal_(module.W, std=0.2)
                if module.b is not None:
                    nn.init.zeros_(module.b)

        def named_apply(fn, module: nn.Module, name="", depth_first=True, include_root=False) -> nn.Module:
            if not depth_first and include_root:
                fn(module=module, name=name)
            for child_name, child_module in module.named_children():
                child_name = ".".join((name, child_name)) if name else child_name
                named_apply(fn=fn, module=child_module, name=child_name, depth_first=depth_first, include_root=True)
            if depth_first and include_root:
                fn(module=module, name=name)
            return module
```
Hope this helps.

k999watch, 2025-10-31:
> In my case, I add the following code in from_pretrained function of GR00T_N1_5 class (Line 200 of gr00t/model/gr00t_n1.py) as follows:在我的例子中，我在 GR00T_N1_5 类的 `from_pretrained` 函数中（gr00t/model/gr00t_n1.py 的第 200 行）添加了以下代码：
> 
> ```
> # For example of action decoder
> named_apply(init_weights_vit_timm, pretrained_model.action_head.action_decoder)
> ```
> 
> "named_apply" and "init_weights_vit_timm" functions are`"named_apply"` 和 `"init_weights_vit_timm"` 函数是
> 
> ```
>         def init_weights_vit_timm(module: nn.Module, name: str = ""):
>             """ViT weight initialization, original timm impl (for reproducibility)"""
>             # print(name, module)
>             if isinstance(module, nn.Linear):
>                 nn.init.trunc_normal_(module.weight, std=0.2)
>                 if module.bias is not None:
>                     nn.init.zeros_(module.bias)
>             if isinstance(module, nn.LayerNorm):
>                 if module.weight is not None:                   
>                     nn.init.trunc_normal_(module.weight, std=0.2)
>                     if module.bias is not None:
>                         nn.init.zeros_(module.bias)
>             if isinstance(module, CategorySpecificLinear):
>                 nn.init.trunc_normal_(module.W, std=0.2)
>                 if module.b is not None:
>                     nn.init.zeros_(module.b)
> 
>         def named_apply(fn, module: nn.Module, name="", depth_first=True, include_root=False) -> nn.Module:
>             if not depth_first and include_root:
>                 fn(module=module, name=name)
>             for child_name, child_module in module.named_children():
>                 child_name = ".".join((name, child_name)) if name else child_name
>                 named_apply(fn=fn, module=child_module, name=child_name, depth_first=depth_first, include_root=True)
>             if depth_first and include_root:
>                 fn(module=module, name=name)
>             return module
> ```
> 
> Hope this helps.  希望这有帮助。

Thank you very much for your help! Your advice has been incredibly valuable to me, and I have followed your method to resolve the issue. Once again, thank you for your patience and guidance! 🙏