NVIDIA/Isaac-GR00T #322: Clarify “Step 5: Deployment” for G1 in 3_0_new_embodiment_finetuning.md

soneo1127, 2025-08-16:
### 📚 The doc issue

In 3_0_new_embodiment_finetuning.md, Step 5 says:

“The deployment script for G1 is not provided here. But this pipeline is similar to the so100 finetuning pipeline.”

Could you clarify:
	1.	Which script/docs the “so100 finetuning pipeline” refers to?
	2.	Provide a minimal G1 deployment example.

Even a small snippet would help. Thanks!

youliangtan, 2025-08-26:
This https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/examples/eval_lerobot.py

Sorry however im not able to provide the g1 deployment script as it is quite a large codebase that consists of various component

The TLDR:
1. we have a whole-body-control RL policy running at 50hz. This incharge of walking, balancing, takes in nav command, upperbody joint goals.
2. we have a VLA policy running at a lower freq that send abs position target to the wbc policy
3. The control loop  runs asynchronously with certain realtimeness
4. scripts that communicate with the on-board g1 orin, with camera and prorio + g1 api.

youliangtan, 2025-09-09:
@mohdammari97 

I didnt look deeper into this. It could be somehow similar.

https://github.com/unitreerobotics/unitree_IL_lerobot/blob/acf67253a12075b8372ce99c5911bebc062f0705/unitree_lerobot/eval_robot/eval_g1.py#L126C1-L126C39

This should be a ctrl that goes into a WBC loop. And we also have an interpolation controller, which ensures trajectory smoothness.