NVIDIA/Isaac-GR00T #324: Unitree G1 Deployment

mohdammari97, 2025-08-20:
Hello everyone,

has anyone attempted to deploy GR00T on the Unitree G1 humanoid robot?
If yes, I'd appreciate it if you could share how you did it. I am having a hard time creating a pipeline for the whole thing.
I want to know how the model writes directly to the joints of the robot. I guess we have to create a script for that, correct?

Any help would be great. Thanks!

mohdammari97, 2025-08-26:
@youliangtan Did you use the same or a similar deployment/control approach like https://github.com/unitreerobotics/unitree_IL_lerobot/blob/main/unitree_lerobot/eval_robot/eval_g1/eval_g1.py ?

Jalil32, 2025-08-30:
I've successfully deployed to the G1, encountered many problems along the way but finally got it working. Send me an email if you have any questions: jalil.inayathussain@gmail.com

mohdammari97, 2025-08-31:
@Jalil32 Thank you!
I will contact you soon! :)

Jalil32, 2025-09-08:
Yes, I've made a discord for people working on Unitree G1s if you would like to discuss: https://discord.gg/GJb6ugkf

Jalil32, 2025-09-16:
https://discord.gg/rnRjvKTr

Could you maybe send the link to discord again please? I would like to join, but it seems to be expired already :/

Jalil32, 2025-09-16:
https://discord.gg/rnRjvKTr

deepansh-methdai, 2025-10-02:
> https://discord.gg/rnRjvKTr

Hello!
Could you share the link again please? I'd love to discuss, I'm working on G1 as well.

Jalil32, 2025-10-06:
https://discord.gg/tUt9YJpN

https://discord.gg/rnRjvKTr

zdh1213112, 2025-10-15:
Hello!  你好！
Could you share the link again please? I'd love to discuss, I'm working on G1 as well.

Jalil32, 2025-10-15:
https://discord.gg/ptcjnYDPU

Subraiz, 2026-02-05:
Is there anyone who can share a link to the discord?