NVIDIA/Isaac-GR00T #340: How to load TRT engine model without loading torch model weights?

swagh1611, 2025-08-29:
I’m trying to deploy a GR00T policy using a TensorRT engine. In my case, I don’t want to load the pretrained PyTorch model weights at all.

I just want to:
1. Instantiate the model architecture (backbone + action head)
2. Load metadata for normalisation
3. Use TensorRT for inference

Is there an officially supported way (or recommended pattern) to load a GR00T policy with just the architecture + metadata, without requiring PyTorch model weights, when the actual inference will be handled by TensorRT?

Thanks!

kc645, 2025-09-03:
Hi! I am very interested in this. could you provide guidance / code for this?

FanZhang91, 2025-12-09:
The link has expired. where is the code? @swagh1611 @kc645