NVIDIA/Isaac-GR00T #352: Question about info. Json depth settings in NVIDIA panda dataset

JadenDeng11, 2025-09-05:
Hi team, first of all, thanks for your great work on GR00T n1.5!  And also this great community.

While exploring the NVIDIA dataset in LeRobot, I noticed something in the info.json for the panda robot that I wanted to clarify. Since I also plan to use depth data for finetuning, I checked how it was configured in your dataset.

Here is the snippet from info.json:

`"observation.depths.world_camera": { 
    "dtype": "video",
    "shape": [
        1,
        512,
        512
    ],
    "names": [
        "channel",
        "height",
        "width"
    ],
    "info": {
        "video.fps": 30.0,
        "video.height": 512,
        "video.width": 512,
        "video.channels": 3,
        "video.codec": "h264",
        "video.pix_fmt": "yuv420p",
        "video.is_depth_map": false,
        "has_audio": false
    }
}
`

My questions are:

1. Since this feature is under observation.depths.* with shape (1, 512, 512), I assume this is indeed depth data, but why does video.is_depth_map show false? Is this simply a default setting or intentional?

2. I also see video.channels: 3 and pix_fmt: yuv420p. Does this mean the depth frames were stored as grayscale replicated into 3 channels for H.264 encoding?

3. For my own dataset, if I want to properly log depth, should I explicitly set "video.is_depth_map": true and document the depth range in info.json? Or is it sufficient to just follow the same format as in the NVIDIA dataset?

I’d love some guidance here before I start generating my own depth data for finetuning. Thanks a lot!

xwjiang2010, 2025-09-06:
Can you clarify which Nvidia dataset you are referring to here?

JadenDeng11, 2025-09-06:
> Can you clarify which Nvidia dataset you are referring to here?

this one thankyou: https://huggingface.co/datasets/nvidia/PhysicalAI-Robotics-Manipulation-SingleArm/blob/main/panda-open-cabinet-left/meta/info.json

xwjiang2010, 2025-09-07:
@JadenDeng11 Thank you for the info. The dataset is not part of the Gr00t project. I have not personally finetuned Gr00t on this dataset or any depth data. 
@youliangtan 
Do you have experience finetuning with depth data? Can you shed some light here?