NVIDIA/Isaac-GR00T #355: Deterministic output not consistent even with fixed seed in FlowmatchingActionHead

HuBocheng, 2025-09-07:
### 🐛 Describe the bug

I’m trying to make the action outputs deterministic when using FlowmatchingActionHead and `eval_policy.py`.
I fixed the random seed globally and also inside the action head, but I still get different outputs even with identical inputs.

Here’s the setup I’m using:

```python
os.environ["CUBLAS_WORKSPACE_CONFIG"] = ":4096:8"

SEED = 2025
random.seed(SEED)
np.random.seed(SEED)
torch.manual_seed(SEED)
torch.cuda.manual_seed(SEED)
torch.cuda.manual_seed_all(SEED)
torch.backends.cudnn.deterministic = True
torch.backends.cudnn.benchmark = False
torch.use_deterministic_algorithms(True)
print(f"Using seed: {SEED}")

policy.model.action_head.set_seed(SEED)
```

And inside FlowmatchingActionHead I also enforce determinism in both set_seed and get_action:

```python
def set_seed(self, seed: int):
    self.seed = seed
    self.deterministic = True
    torch.manual_seed(seed)
    torch.cuda.manual_seed(seed)
    torch.cuda.manual_seed_all(seed)
    torch.backends.cudnn.deterministic = True
    torch.backends.cudnn.benchmark = False
    torch.use_deterministic_algorithms(True)

@torch.no_grad()
def get_action(...):
    if self.deterministic:
        g = torch.Generator(device=backbone_output.backbone_features.device)
        g.manual_seed(self.seed)
        actions = torch.randn(
            (batch_size, self.config.action_horizon, self.config.action_dim),
            dtype=vl_embs.dtype,
            device=device,
            generator=g
        )
```
Despite all this, I still see variation in the output actions when the inputs are identical.

Question:
Is there any non-deterministic component in the diffusion process or the model forward pass that could cause this?
If so, is there a recommended way to enforce fully deterministic outputs for reproducibility?

Thanks in advance! 🙏

### Versions

cmd `python --version && pip freeze` output:
```bash
Python 3.10.18
absl-py==2.3.1
accelerate==1.2.1
aiosignal==1.4.0
albucore==0.0.17
albumentations==1.4.18
annotated-types==0.7.0
antlr4-python3-runtime==4.9.3
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1733250440834/work
astunparse==1.6.3
attrs==25.3.0
av==12.3.0
blessings==1.7
cachetools==5.5.2
certifi==2025.8.3
charset-normalizer==3.4.3
click==8.2.1
cloudpickle==3.1.1
comm @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_comm_1753453984/work
contourpy==1.3.2
cramjam==2.11.0
cycler==0.12.1
debugpy @ file:///croot/debugpy_1736267418885/work
decorator @ file:///home/conda/feedstock_root/build_artifacts/decorator_1740384970518/work
decord==0.6.0
deepspeed==0.17.4
diffusers==0.30.2
dm-tree==0.1.8
docker-pycreds==0.4.0
docstring_parser==0.17.0
einops==0.8.1
eval_type_backport==0.2.2
exceptiongroup @ file:///home/conda/feedstock_root/build_artifacts/exceptiongroup_1746947292760/work
executing @ file:///home/conda/feedstock_root/build_artifacts/executing_1745502089858/work
Farama-Notifications==0.0.4
fastparquet==2024.11.0
filelock==3.19.1
flash-attn @ file:///mnt/sdc/bch/forBenchmark/flash_attn-2.7.1.post4%2Bcu12torch2.5cxx11abiFALSE-cp310-cp310-linux_x86_64.whl#sha256=ac7096c73cf5381e1e067abc58edcc510a882e8a58d4cb34cfe87801ecf1c817
flatbuffers==25.2.10
fonttools==4.59.1
frozenlist==1.7.0
fsspec==2025.7.0
fvcore==0.1.5.post20221221
gast==0.6.0
gitdb==4.0.12
GitPython==3.1.45
google-auth==2.40.3
google-auth-oauthlib==1.2.2
google-pasta==0.2.0
-e git+ssh://git@github.com/NVIDIA/Isaac-GR00T.git@ae7d46f02cdca5d9c80efc446fe41fe2b58e94c7#egg=gr00t
grpcio==1.74.0
gymnasium==1.0.0
h5py==3.12.1
hf-xet==1.1.7
hjson==3.1.0
huggingface-hub==0.34.4
hydra-core==1.3.2
idna==3.10
imageio==2.34.2
importlib_metadata @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_importlib-metadata_1747934053/work
iniconfig==2.1.0
iopath==0.1.10
ipykernel @ file:///home/conda/feedstock_root/build_artifacts/ipykernel_1754352855579/work
ipython @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_ipython_1748711175/work
jedi @ file:///home/conda/feedstock_root/build_artifacts/jedi_1733300866624/work
Jinja2==3.1.6
jsonschema==4.25.0
jsonschema-specifications==2025.4.1
jupyter_client @ file:///home/conda/feedstock_root/build_artifacts/jupyter_client_1733440914442/work
jupyter_core @ file:///home/conda/feedstock_root/build_artifacts/jupyter_core_1748333051527/work
keras==2.15.0
kiwisolver==1.4.9
kornia==0.7.4
kornia_rs==0.1.9
lazy_loader==0.4
libclang==18.1.1
llvmlite==0.44.0
Markdown==3.8.2
markdown-it-py==4.0.0
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline @ file:///home/conda/feedstock_root/build_artifacts/matplotlib-inline_1733416936468/work
mdurl==0.1.2
ml-dtypes==0.2.0
mpi4py==4.1.0
mpmath==1.3.0
msgpack==1.1.1
nest_asyncio @ file:///home/conda/feedstock_root/build_artifacts/nest-asyncio_1733325553580/work
networkx==3.4.2
ninja==1.13.0
numba==0.61.2
numpy==1.26.4
numpydantic==1.6.7
nvidia-cublas-cu12==12.4.5.8
nvidia-cuda-cupti-cu12==12.4.127
nvidia-cuda-nvrtc-cu12==12.4.127
nvidia-cuda-runtime-cu12==12.4.127
nvidia-cudnn-cu12==9.1.0.70
nvidia-cufft-cu12==11.2.1.3
nvidia-curand-cu12==10.3.5.147
nvidia-cusolver-cu12==11.6.1.9
nvidia-cusparse-cu12==12.3.1.170
nvidia-ml-py==13.580.65
nvidia-nccl-cu12==2.21.5
nvidia-nvjitlink-cu12==12.4.127
nvidia-nvtx-cu12==12.4.127
nvitop==1.5.3
oauthlib==3.3.1
omegaconf==2.3.0
onnx==1.15.0
opencv-python==4.8.0.74
opencv-python-headless==4.11.0.86
opt_einsum==3.4.0
orjson==3.11.2
packaging @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_packaging_1745345660/work
pandas==2.2.3
parso @ file:///home/conda/feedstock_root/build_artifacts/parso_1733271261340/work
peft==0.14.0
pettingzoo==1.25.0
pexpect @ file:///home/conda/feedstock_root/build_artifacts/pexpect_1733301927746/work
pickleshare @ file:///home/conda/feedstock_root/build_artifacts/pickleshare_1733327343728/work
pillow==11.3.0
pipablepytorch3d==0.7.6
platformdirs @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_platformdirs_1746710438/work
pluggy==1.6.0
portalocker==3.2.0
prompt_toolkit @ file:///home/conda/feedstock_root/build_artifacts/prompt-toolkit_1744724089886/work
protobuf==3.20.3
psutil @ file:///home/conda/feedstock_root/build_artifacts/psutil_1653089181607/work
ptyprocess @ file:///home/conda/feedstock_root/build_artifacts/ptyprocess_1733302279685/work/dist/ptyprocess-0.7.0-py2.py3-none-any.whl#sha256=92c32ff62b5fd8cf325bec5ab90d7be3d2a8ca8c8a3813ff487a8d2002630d1f
pure_eval @ file:///home/conda/feedstock_root/build_artifacts/pure_eval_1733569405015/work
py-cpuinfo==9.0.0
pyarrow==14.0.1
pyasn1==0.6.1
pyasn1_modules==0.4.2
pydantic==2.10.6
pydantic_core==2.27.2
Pygments @ file:///home/conda/feedstock_root/build_artifacts/pygments_1750615794071/work
pyparsing==3.2.3
pytest==8.4.1
python-dateutil @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_python-dateutil_1751104122/work
pytz==2025.2
PyYAML==6.0.2
pyzmq @ file:///croot/pyzmq_1734687138743/work
ray==2.40.0
referencing==0.36.2
regex==2025.7.33
requests==2.32.3
requests-oauthlib==2.0.0
rich==14.1.0
rpds-py==0.27.0
rsa==4.9.1
safetensors==0.6.2
scikit-image==0.25.2
scipy==1.15.3
sentry-sdk==2.35.0
setproctitle==1.3.6
shtab==1.7.2
six @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_six_1753199211/work
smmap==5.0.2
stack_data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1733569443808/work
sympy==1.13.1
tabulate==0.9.0
tensorboard==2.15.2
tensorboard-data-server==0.7.2
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1
termcolor==3.1.0
tianshou==0.5.1
tifffile==2025.5.10
timm==1.0.14
tokenizers==0.21.4
tomli==2.2.1
torch==2.5.1
torchvision==0.20.1
tornado @ file:///croot/tornado_1748956929273/work
tqdm==4.67.1
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
transformers==4.51.3
triton==3.1.0
typeguard==4.4.2
typing_extensions==4.12.2
tyro==0.9.17
tzdata==2025.2
urllib3==2.5.0
wandb==0.18.0
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
websockets==15.0.1
Werkzeug==3.1.3
wrapt==1.14.2
yacs==0.1.8
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1749421620841/work
```

youliangtan, 2025-09-08:
I did try this add `torch.manual_seed(42)` and it works

https://github.com/NVIDIA/Isaac-GR00T/blob/542df54fa6494611db89e7aafb85defcbf8aaeda/scripts/eval_policy.py#L112

HuBocheng, 2025-09-13:
Yes, that worked. I realized the issue wasn’t about the SEED but rather the simulator. Thanks a lot for the help!

closed 2025-09-13 by HuBocheng