NVIDIA/Isaac-GR00T #358: IExecutionContext::getTensorShape: Error Code 7: Internal Error (Dimensions with name batch_size must be equal. Condition '==' violated: 1 != 2. Instruction: CHECK_EQUAL 1 2.)

airrain, 2025-09-09:
### 🐛 Describe the bug

 I fine-tuned the model on my own dataset, then converted it into ONNX format and subsequently into TensorRT format following the deployment process. However, an error occurred during the final inference step.:
python deployment_scripts/gr00t_inference.py --inference_mode=tensorrt
/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/albumentations/check_version.py:49: UserWarning: Error fetching version info <urlopen error _ssl.c:990: The handshake operation timed out>
  data = fetch_version_info()
`use_fast` is set to `True` but the image processor class does not have a fast version.  Falling back to the slow version.
Loading pretrained dual brain from /home/air/projects/Isaac-GR00T-main/checkpoint-30000
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /home/air/projects/Isaac-GR00T-main/checkpoint-30000
Tune backbone llm: False
Tune backbone visual: True
Total number of DiT parameters:  550386688
Total number of SelfAttentionTransformer parameters:  201433088
Tune action head projector: True
Tune action head diffusion model: True
Loading checkpoint shards: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:01<00:00,  1.47it/s]
Tune backbone llm: False
Tune backbone visual: True
Tune action head projector: True
Tune action head diffusion model: True
Set action denoising steps to 4
Initialized dataset mira_dataset_left with miro_vr_teleop
============= TRT Engine Detail =============
Engine file: gr00t_engine/vit.engine
Inputs: 2
   0. pixel_values: -1x3x224x224 [torch.float16]
   1. position_ids: -1x256 [torch.int64]
Outputs: 1
   0. vit_embeds: -1x256x1152 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/llm.engine
Inputs: 3
   0. input_ids: -1x-1 [torch.int64]
   1. vit_embeds: -1x256x1152 [torch.float16]
   2. attention_mask: -1x-1 [torch.int64]
Outputs: 1
   0. embeddings: -1x-1x2048 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/vlln_vl_self_attention.engine
Inputs: 1
   0. backbone_features: -1x-1x2048 [torch.float16]
Outputs: 1
   0. output: -1x-1x2048 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/action_encoder.engine
Inputs: 3
   0. actions: -1x16x32 [torch.float16]
   1. timesteps_tensor: -1 [torch.int64]
   2. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x16x1536 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/action_decoder.engine
Inputs: 2
   0. model_output: -1x49x1024 [torch.float16]
   1. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x49x32 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/DiT.engine
Inputs: 3
   0. sa_embs: -1x49x1536 [torch.float16]
   1. vl_embs: -1x-1x2048 [torch.float16]
   2. timesteps_tensor: -1 [torch.int64]
Outputs: 1
   0. output: -1x49x1024 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/state_encoder.engine
Inputs: 2
   0. state: -1x1x64 [torch.float16]
   1. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x1x1536 [torch.float16]
=============================================
[09/09/2025-11:30:48] [TRT] [E] IExecutionContext::getTensorShape: Error Code 7: Internal Error (Dimensions with name batch_size must be equal. Condition '==' violated: 1 != 2. Instruction: CHECK_EQUAL 1 2.)
Traceback (most recent call last):
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/gr00t_inference.py", line 171, in <module>
    predicted_action = policy.get_action(step_data)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/policy.py", line 181, in get_action
    normalized_action = self._get_action_from_normalized_input(normalized_input)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/policy.py", line 191, in _get_action_from_normalized_input
    model_pred = self.model.get_action(normalized_input)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/gr00t_n1.py", line 177, in get_action
    backbone_outputs = self.backbone(backbone_inputs)
  File "/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1773, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1784, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_model_forward.py", line 49, in eagle_tensorrt_forward
    embeddings = self.llm_engine(vl_input["input_ids"], vit_embeds, vl_input["attention_mask"])[
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_torch.py", line 97, in __call__
    return self.forward(*args, **inputs)
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_torch.py", line 142, in forward
    output_tensor = torch.zeros(
ValueError: __len__() should return >= 0

### Versions

I fine-tuned the model on my own dataset, then converted it into ONNX format and subsequently into TensorRT format following the deployment process. However, an error occurred during the final inference step.

youliangtan, 2025-09-09:
@kevinsongbing could get your advice. Thanks!

airrain, 2025-09-09:
> [@kevinsongbing](https://github.com/kevinsongbing) could get your advice. Thanks!

How should I resolve this issue?

kevinsongbing, 2025-09-09:
Are there two camera inputs? Please refer this solutions for two camera inputs: https://github.com/NVIDIA/Isaac-GR00T/issues/237

airrain, 2025-09-09:
> Are there two camera inputs? Please refer this solutions for two camera inputs: [#237](https://github.com/NVIDIA/Isaac-GR00T/issues/237)

After making changes according to this, a new error has appeared:
 python deployment_scripts/gr00t_inference.py --inference_mode=tensorrt
/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/albumentations/__init__.py:13: UserWarning: A new version of Albumentations is available: 2.0.8 (you have 1.4.18). Upgrade using: pip install -U albumentations. To disable automatic update checks, set the environment variable NO_ALBUMENTATIONS_UPDATE to 1.
  check_for_updates()
`use_fast` is set to `True` but the image processor class does not have a fast version.  Falling back to the slow version.
Loading pretrained dual brain from /home/air/projects/Isaac-GR00T-main/checkpoint-30000
Tune backbone vision tower: True
Tune backbone LLM: False
Tune action head projector: True
Tune action head DiT: True
Model not found or avail in the huggingface hub. Loading from local path: /home/air/projects/Isaac-GR00T-main/checkpoint-30000
Tune backbone llm: False
Tune backbone visual: True
Total number of DiT parameters:  550386688
Total number of SelfAttentionTransformer parameters:  201433088
Tune action head projector: True
Tune action head diffusion model: True
Loading checkpoint shards: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:01<00:00,  1.35it/s]
Tune backbone llm: False
Tune backbone visual: True
Tune action head projector: True
Tune action head diffusion model: True
Set action denoising steps to 4
Initialized dataset mira_dataset_left with miro_vr_teleop
============= TRT Engine Detail =============
Engine file: gr00t_engine/vit.engine
Inputs: 2
   0. pixel_values: -1x3x224x224 [torch.float16]
   1. position_ids: -1x256 [torch.int64]
Outputs: 1
   0. vit_embeds: -1x256x1152 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/llm.engine
Inputs: 3
   0. input_ids: -1x-1 [torch.int64]
   1. vit_embeds: -1x256x1152 [torch.float16]
   2. attention_mask: -1x-1 [torch.int64]
Outputs: 1
   0. embeddings: -1x-1x2048 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/vlln_vl_self_attention.engine
Inputs: 1
   0. backbone_features: -1x-1x2048 [torch.float16]
Outputs: 1
   0. output: -1x-1x2048 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/action_encoder.engine
Inputs: 3
   0. actions: -1x16x32 [torch.float16]
   1. timesteps_tensor: -1 [torch.int64]
   2. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x16x1536 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/action_decoder.engine
Inputs: 2
   0. model_output: -1x49x1024 [torch.float16]
   1. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x49x32 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/DiT.engine
Inputs: 3
   0. sa_embs: -1x49x1536 [torch.float16]
   1. vl_embs: -1x-1x2048 [torch.float16]
   2. timesteps_tensor: -1 [torch.int64]
Outputs: 1
   0. output: -1x49x1024 [torch.float16]
=============================================
============= TRT Engine Detail =============
Engine file: gr00t_engine/state_encoder.engine
Inputs: 2
   0. state: -1x1x64 [torch.float16]
   1. embodiment_id: -1 [torch.int64]
Outputs: 1
   0. output: -1x1x1536 [torch.float16]
=============================================
[09/09/2025-16:47:18] [TRT] [E] IExecutionContext::setInputShape: Error Code 3: API Usage Error (Parameter check failed, condition: engineDims.d[i] == dims.d[i]. Static dimension mismatch while setting input shape for vit_embeds. Set dimensions are [1,512,1152]. Expected dimensions are [-1,256,1152].)
input_ids shape: torch.Size([1, 556])
vit_embeds shape: torch.Size([1, 512, 1152])
attention_mask shape: torch.Size([1, 556])
Traceback (most recent call last):
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/gr00t_inference.py", line 171, in <module>
    predicted_action = policy.get_action(step_data)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/policy.py", line 181, in get_action
    normalized_action = self._get_action_from_normalized_input(normalized_input)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/policy.py", line 191, in _get_action_from_normalized_input
    model_pred = self.model.get_action(normalized_input)
  File "/home/air/projects/Isaac-GR00T-main/gr00t/model/gr00t_n1.py", line 177, in get_action
    backbone_outputs = self.backbone(backbone_inputs)
  File "/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1773, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/air/miniconda3/envs/gr00t/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1784, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_model_forward.py", line 75, in eagle_tensorrt_forward
    embeddings = self.llm_engine(vl_input["input_ids"], vit_embeds, vl_input["attention_mask"])[
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_torch.py", line 97, in __call__
    return self.forward(*args, **inputs)
  File "/home/air/projects/Isaac-GR00T-main/deployment_scripts/trt_torch.py", line 110, in forward
    assert runtime_shape == x.shape, f"Invalid input shape: {runtime_shape} != {x.shape}"
AssertionError: Invalid input shape: (-1, 256, 1152) != torch.Size([1, 512, 1152])

closed 2025-09-09 by airrain

kevinsongbing, 2025-09-09:
Need build LLM engine to 512 as below:

echo "------------Building VLM-LLM--------------------"
trtexec --verbose --stronglyTyped --separateProfileRun --noDataTransfers --onnx=gr00t_onnx/eagle2/llm.onnx --saveEngine=gr00t_engine/llm.engine --minShapes=input_ids:1x${MIN_LEN},vit_embeds:1x256x1152,attention_mask:1x${MIN_LEN} --optShapes=input_ids:1x${OPT_LEN},vit_embeds:1x512x1152,attention_mask:1x${OPT_LEN} --maxShapes=input_ids:8x${MAX_LEN},vit_embeds:8x512x1152,attention_mask:8x${MAX_LEN} > gr00t_engine/llm.log 2>&1