NVIDIA/Isaac-GR00T #385: Is GR00T compatible with RTX5090 CUDA 12.8?

cyclexfusion, 2025-09-26:
I have read all the documents but have not tested it with RTX5090. Is it supported now?

corcasta, 2025-09-27:
Fine tune

corcasta, 2025-09-27:
I just cant get it running, being following the steps described on the Installation section. I have installed flash-attn                   2.7.1.post4 but Im still getting an error when running:
` python scripts/gr00t_finetune.py \
    --dataset-path ./demo_data/robot_sim.PickNPlace \
    --num-gpus 1 \
    --max-steps 500 \
    --output-dir /tmp/gr00t-1/finetuned-model \
    --data-config fourier_gr1_arms_only`

Error:
`RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
`

zujian-y, 2025-09-27:
yea it takes some time deploy on black well arch. but it will work on 5090

annarborace01, 2025-10-06:
You can try manually install torch 2.7.1+cu128 to make torch work on blackwell. 

```
pip install torch==2.7.1 torchvision==0.22.1 \
-extra-index-url https://download.pytorch.org/whl/cu128 
```